arendst/Tasmota #2544: Wifi reconnect uses a lot of Heap

SecurityAddicted, 2018-04-25:
I have a Sonoff which is quite far from the AP and frequently drops the connection and try to reconnect (wificonfig is set to 5). What I noticed is after a lot of retries the remaining heap starts to go down, until the device becomes unreachable (probably crashed).
Any thoughts about that?

Using Tasmota v5.12 with SDK 2.4.1.

arendst, 2018-04-25:
What LwIP option did you use when compiling with 2.4.1?

The most obvious answer is not to use 2.4.1 as it is known to have memory leaks.

SecurityAddicted, 2018-04-25:
Using v2 Lower Memory. What do you suggest? Differences about them?

arendst, 2018-04-25:
You might want to try the latest development version which contains a fix for this memory leak as documented here https://github.com/esp8266/Arduino/issues/4497

So re-compile with  V2 lower memory and 2.4.1 and let me know the results over time.

ascillato, 2018-06-04:
Hi,

If your issue was solved, please, close it. Thanks :+1:

closed 2018-06-12 by ascillato2