arendst/Tasmota #2547: SI7021 or AM2301? suggestion

vedatorkun, 2018-04-25:
Which sensor should I buy for sonoff th16 tasmota?
SI7021 or AM2301? Also what is the difference between them ?, please help me.

Frogmore42, 2018-04-26:
http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html

vedatorkun, 2018-04-26:
Thank you for the article link. I just want to measure the room temperature and humidity as the price / performance maximum. Of course do should not cause problems in tasmota process.

Frogmore42, 2018-04-26:
What are you doing with the data?
How accurate do you want it to be?

The cheap sensors are inexpensive and not at all accurate.

I like the BME280.  It is reasonably priced and appears quite accurate.

vedatorkun, 2018-04-26:
I will use it with my home assistant (hass.io). My target is just to learn room temperature and humibility. They sell these two models together with the "sonoff th16" device in aliexpress. I want tasmota to be faultless.

closed 2018-06-25 by stale[bot]