arendst/Tasmota #2548: Rule publish only after POWER1/2 state OFF

mobiipl, 2018-04-25:
Sonoff Dual (tasmota 5.12.0l)

I want to publish message every POWER (1 or 2) change from ON to OFF and opposite
Device publish message from rule only after POWER state going to OFF and send both messages (UP/STOP or DOWN/STOP) in this same time. When POWER state going to ON nothing heppen. Maybe I have mistake in rule or don't understanding scheme.

Rule:
`on POWER1#State=ON do publish stat/roleta/stan UP endon on POWER1#State=OFF do publish stat/roleta/stan STOP endon on POWER2#State=ON do publish stat/roleta/stan DOWN endon on POWER2#State=OFF do publish stat/roleta/stan STOP endon`

Console:
22:05:41 MQT: stat/roleta/RESULT = {"POWER1":"ON"}
22:05:41 MQT: stat/roleta/POWER1 = ON
22:05:47 RUL: POWER1#STATE=ON performs "publish stat/roleta/stan UP"
22:05:47 MQT: stat/roleta/stan = UP
22:05:47 MQT: stat/roleta/RESULT = {"Publish":"Done"}
22:05:47 RUL: POWER1#STATE=OFF performs "publish stat/roleta/stan STOP"
22:05:47 MQT: stat/roleta/stan = STOP
22:05:47 MQT: stat/roleta/RESULT = {"Publish":"Done"}
22:05:47 MQT: stat/roleta/RESULT = {"POWER1":"OFF"}
22:05:47 MQT: stat/roleta/POWER1 = OFF
22:05:51 MQT: stat/roleta/RESULT = {"POWER2":"ON"}
22:05:51 MQT: stat/roleta/POWER2 = ON
22:05:56 RUL: POWER2#STATE=ON performs "publish stat/roleta/stan DOWN"
22:05:56 MQT: stat/roleta/stan = DOWN
22:05:56 MQT: stat/roleta/RESULT = {"Publish":"Done"}
22:05:56 RUL: POWER2#STATE=OFF performs "publish stat/roleta/stan STOP"
22:05:56 MQT: stat/roleta/stan = STOP
22:05:56 MQT: stat/roleta/RESULT = {"Publish":"Done"}
22:05:56 MQT: stat/roleta/RESULT = {"POWER2":"OFF"}
22:05:56 MQT: stat/roleta/POWER2 = OFF

Log OpenHAB:
`2018-04-25 23:05:47.278 [vent.ItemStateChangedEvent] - Sypialnia_Roleta_Stan changed from STOP to UP
2018-04-25 23:05:47.285 [vent.ItemStateChangedEvent] - Sypialnia_Roleta_Stan changed from UP to STOP
2018-04-25 23:05:56.405 [vent.ItemStateChangedEvent] - Sypialnia_Roleta_Stan changed from STOP to DOWN
2018-04-25 23:05:56.422 [vent.ItemStateChangedEvent] - Sypialnia_Roleta_Stan changed from DOWN to STOP`

arendst, 2018-04-26:
You might want to try POWER1#State=1 instead of ON and POWER1#State=0 instead of OFF

mobiipl, 2018-04-26:
Thanks for solution, I came to this yesterday after some time, and work :) Maybe it would be good idea to add ON/OFF possibility?

timtimsson, 2018-04-27:
Hi,
sorry for this stupid question... how is the syntax in the user_config_override.h

only ?
`on POWER1#State=1 do publish stat/roleta/stan UP endon`
or
```
// Examples
#ifdef CFG_HOLDER
#undef CFG_HOLDER
#endif
#define CFG_HOLDER   0x20170324

#define USE_RULES      "on POWER1#State=1 do publish stat/roleta/stan UP endon"
```
or?

closed 2018-04-29 by arendst