arendst/Tasmota #2549: What is the "Once" setting in rules? How do I use it?

thenameiwantedwastaken, 2018-04-26:
I just started using rules, and noticed that I get:

RESULT = {"Rule":"ON","Once":"OFF","Rules":"on ENERGY#Current>0.100 do p... 

What is the "Once" value for? And how do I change it? I didn't see anything about it in the Wiki.

Thanks

thenameiwantedwastaken, 2018-04-26:
Answer found in xdrv_10_rules.ino:

```
 *   Control rule triggering with command:
 *     Rule 0 = Rules disabled (Off)
 *     Rule 1 = Rules enabled (On)
 *     Rule 2 = Toggle rules state
 *     Rule 4 = Perform commands as long as trigger is met (Once OFF)
 *     Rule 5 = Perform commands once until trigger is not met (Once ON)
 *     Rule 6 = Toggle Once state
 *   Execute an event like:
 *     Event anyname=001000
 *   Set a RuleTimer to 100 seconds like:
 *     RuleTimer2 100
```

An example:
`RULE on ENERGY#Current>0.100 do publish tool/tablesaw/power 1 endon on ENERGY#Current<0.100 do publish tool/tablesaw/power 0 endon`
This creates a rule to publish MQTT commands whenever power is on or off. Used as as, it will publish MQTT commands repeatedly, over and over... but, add another command:
`Rule 5`
Now the MQTT message will be sent once, and only once, while the condition is met. This is perfect for thermostat on/off depending on temperature, bathroom extractor fan on/off depending on humidity, workshop dust collector on/off depending on whether some machine is running.

I hope this helps.

arendst, 2018-04-26:
Thnx. Updated the wiki with your findings.

closed 2018-04-26 by arendst

tbagro, 2018-08-10:
I understood how to assemble the rules, but in which function are the rules inserted?

rasti70, 2019-01-10:
I want to setup a Blitzwolf SHP2 / SHP6 wifi power measuring  plug
in a way that is does not publish  MQTT messages if power is less than 5W.

If Power>5W only peridic teleperiod MQTT messages should be sent out.
MQTT topic is "Mess_Steckdose_99"

rule1 on ENERGY#Power>5 do publish tele/Mess_Steckdose_99/SENSOR  endon
does not work

Can you please explain how the rule I need should be implemented ?

Thanks !

Ralf

meingraham, 2019-01-10:
Take a look at the Commands wiki sections on `PowerLow`, `PowerHigh`, and `PowerDelta`. Also, make sure `SetOption21` is set appropriately.

rasti70, 2019-01-10:
Thank you Micheal !

I am using Powerdelta. It works correctly (tried with variable load / soldering/iron).
PowerLow is exactly what I need, however, seems it does not work.
I set Powerlow 10 and nom matter wether I set SetOption21 to 0 or to 1 
tasmota produces messages each 2-3 seconds.