arendst/Tasmota #2551: WiFi connection failure

arturodiazad1, 2018-04-26:
hi, all the basic Sonoff in which install TASMOTA with 5.10 perfect but those that I have recently installed with the version 5.12 only Connect with the Wi-Fi on some occasions, I do not know what to do, I check it in the Arduino terminal and continuously give connection failure, has someone happened to me or can you help me solve it?

Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

reloxx13, 2018-04-26:
Provide the output of command status 0

MatzeJoerling, 2018-04-26:
what about reading the wiki, perhaps the commands section:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

So, open your console (serial, web, mqtt also possible) and type "status 0"
Post the output!

arturodiazad1, 2018-04-27:
This is what happens, the module does not connect to the Wi-Fi network, only in some cases it connects to the network. This is suspending me since downloading version 5.12 all the accessories that are working with vrsion 5.10 work perfect.

[image]

arturodiazad1, 2018-04-27:
You can't comment at this time — your comment cannot be blank.

[image]

Frogmore42, 2018-04-29:
You might try using core 2.3.0 instead of the 2.4.0 you are currently using.  Some people have had connection problems with 2.4.0.

titomontes, 2018-04-30:
I know what question comes next: How do I get to change the Core version?? before flashing?

Thanks

arturodiazad1, 2018-05-01:
update the libraries to the latest version and the connection problems have been solved. thank you very much

closed 2018-05-01 by arturodiazad1