arendst/Tasmota #2552: OTA with default url work, but not with my own

bollitec, 2018-04-26:
Hello there,

hopefully you can help.

I use the default OTA release change without SonOTA. For the url: http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin works it perfect.

My issue is:
If I use my own url, like: http://raspi/sonoffs/sonoff.bin the 404 (File not found error) happens.
The file with each webbrowser downlodable and have 775 for the access.

My Workflows:

a) works well:
     1. mosquitto_pub -t 'cmnd/sonoffs/Backlog' -m 'otaurl http://sonoff.maddox.co.uk/tasmota/sonoff.bin'
     2. mosquitto_pub -t 'cmnd/sonoffs/Backlog' -m 'upgrade 1'

b) doesn't work => 404:
   1. mosquitto_pub -t 'cmnd/sonoffs/Backlog' -m 'otaurl http://raspi/sonoffs/sonoff.bin'
   2. mosquitto_pub -t 'cmnd/sonoffs/Backlog' -m 'upgrade 1'

Any idea?
Thanks in advanced,
Jörg

MatzeJoerling, 2018-04-26:
put in the ip of your raspi! if that works fix your dns setup.

regards M.

bollitec, 2018-04-26:
done this already, isn't work as well.

Have something in mind and check this => file size ... there are only 516kb free and 530kb is to much

reloxx13, 2018-04-26:
> put in the ip of your raspi!

this.

ans yep, if the file is too big, it wont upgrade and just restart.

MatzeJoerling, 2018-04-26:
can you post the output of the "curl" command?

bollitec, 2018-04-28:
@all,

test some time and found now the issue as before named: the file_size
It checks if file exist, then if the size able to flash and if something going wrong it tries to load the "sonoff-minimal.bin". This one i don't had stored on my own web server => 404 File not found.

So my issue is fixed with more space at wemos/nodemcu with add new esp8266.flash.4m1m.ld and change at the platformio.ini:
              board = nodemcuv2
              build_flags = -Wl,-Tesp8266.flash.4m1m.ld
After upload 2500kb were available and 404 doesn't happen anymore :)

Many thanks to all!!!

esp8266.flash.4m1m.ld (partly copied from the eagle file, change last line):
/* Flash Split for 4M chips */
/* sketch 1019KB */
/* empty  2048KB */
/* spiffs 1004KB */
/* eeprom 20KB */

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xf9ff0
}

PROVIDE ( _SPIFFS_start = 0x40500000 );
PROVIDE ( _SPIFFS_end = 0x405FB000 );
PROVIDE ( _SPIFFS_page = 0x100 );
PROVIDE ( _SPIFFS_block = 0x2000 );

INCLUDE "esp8266.flash.common.ld"

closed 2018-04-28 by bollitec