arendst/Tasmota #2553: Platform.io: Build definition platformio.ini => board?!?

bollitec, 2018-04-26:
Hey there,

please a question to the platfomio.ini regarding the defined board.
Actual are all defined with => **board = esp01_1m**
Like now to use Wemos Mini and NodeMCU with 4MB storage.

What is the correct board I should define there?

Thanks a lot,
Jörg

reloxx13, 2018-04-26:
Example:
```
[env:sonoff-DE-4MB]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board}
board_flash_mode = ${common.board_flash_mode}
build_unflags = ${common.build_unflags}
build_flags =
  -Wl,-Tesp8266.flash.4m1m.ld
  -DVTABLES_IN_FLASH
  -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
  -DUSE_CONFIG_OVERRIDE
  -DMY_LANGUAGE=de-DE
monitor_baud = ${common.monitor_baud}
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
extra_scripts = ${common.extra_scripts}
```

`  -Wl,-Tesp8266.flash.4m1m.ld`

arendst, 2018-04-27:
I only support esp01_1m even when you think you have more flash available. 

You are responsible for any OTA error or setting loss that may result in mixing board types.

closed 2018-04-28 by bollitec