arendst/Tasmota #2557: OLED Display support

rlust, 2018-04-27:
Checking to see if any display support is planned for the future?
I see great value in using OLED displays for non SONOFF devices, like WEMOS.

arendst, 2018-04-28:
I have some display support running for over a year but it is very user unfriendly as it only allows displaying of MQTT messages without user intervention. Displays supported are LCD, Oled, TFT and up to 8 matrices. AT home they run fine.

I might release a branch without support just for you to see what is possible.

rlust, 2018-04-30:
Got it loaded, not sure has to set options?
I am using a IC2 old display.
Anyone have any luck here?

arendst, 2018-04-30:
Currently all displays have the possibility to show life MQTT messages. This probably only works when the default prefixes are used. The display driver receives all tele messages and decodes the JSON text and should display information like

[image]
Settings:
```
14:39:16 CMD: displaymodel
14:39:16 MQT: stat/wemos9/RESULT = {"DisplayModel":2}
14:39:20 CMD: displaymode
14:39:20 MQT: stat/wemos9/RESULT = {"DisplayMode":5}
14:39:33 CMD: displayrefresh
14:39:33 MQT: stat/wemos9/RESULT = {"DisplayRefresh":2}
14:39:45 CMD: power
14:39:45 MQT: stat/wemos9/RESULT = {"POWER":"ON"}
14:39:45 MQT: stat/wemos9/POWER = ON
14:39:56 CMD: displayrows
14:39:56 MQT: stat/wemos9/RESULT = {"DisplayRows":8}
14:40:01 CMD: displaycols
14:40:01 MQT: stat/wemos9/RESULT = {"DisplayCols1":21}
```
[image]
Settings:
```
15:41:23 CMD: displaymodel
15:41:23 MQT: stat/ticker/RESULT = {"DisplayModel":3}
15:41:28 CMD: displaymode
15:41:28 MQT: stat/ticker/RESULT = {"DisplayMode":5}
15:41:34 CMD: displayrefresh
15:41:34 MQT: stat/ticker/RESULT = {"DisplayRefresh":1}
15:41:38 CMD: power
15:41:38 MQT: stat/ticker/RESULT = {"POWER":"ON"}
15:41:38 MQT: stat/ticker/POWER = ON
15:41:49 CMD: displayrows
15:41:49 MQT: stat/ticker/RESULT = {"DisplayRows":2}
15:41:54 CMD: displaycols
15:41:54 MQT: stat/ticker/RESULT = {"DisplayCols1":32}
```
[image]
Settings:
```
15:42:46 CMD: displaymodel
15:42:46 MQT: stat/lcd/RESULT = {"DisplayModel":1}
15:42:50 CMD: displaymode
15:42:50 MQT: stat/lcd/RESULT = {"DisplayMode":4}
15:42:58 CMD: displayrefresh
15:42:58 MQT: stat/lcd/RESULT = {"DisplayRefresh":2}
15:43:01 CMD: power
15:43:01 MQT: stat/lcd/RESULT = {"POWER":"ON"}
15:43:01 MQT: stat/lcd/POWER = ON
15:43:07 CMD: displayrows
15:43:07 MQT: stat/lcd/RESULT = {"DisplayRows":2}
15:43:11 CMD: displaycols
15:43:11 MQT: stat/lcd/RESULT = {"DisplayCols1":16}
15:43:24 CMD: displayaddress
15:43:24 MQT: stat/lcd/RESULT = {"DisplayAddress1":39}
```
[image]
Settings:
```
15:44:10 CMD: displaymodel
15:44:10 MQT: stat/tft/RESULT = {"DisplayModel":4}
15:44:14 CMD: displaymode
15:44:14 MQT: stat/tft/RESULT = {"DisplayMode":5}
15:44:21 CMD: displayrefresh
15:44:21 MQT: stat/tft/RESULT = {"DisplayRefresh":2}
15:44:24 CMD: power
15:44:24 MQT: stat/tft/RESULT = {"POWER":"ON"}
15:44:24 MQT: stat/tft/POWER = ON
15:44:28 CMD: displayrows
15:44:28 MQT: stat/tft/RESULT = {"DisplayRows":2}
15:44:34 CMD: displaycols
15:44:34 MQT: stat/tft/RESULT = {"DisplayCols1":20}
```

Jason2866, 2018-04-30:
THANK you!
More explenations as hoped!
With that the last open point in my Smart Home project is going to be solved.
I need a few simple and cheap devices that displays some parameters (temperature, humidity)😃

rlust, 2018-04-30:
Thanks so much Arendst, your work is invaluable!
Well done as always!

Jason2886, are you able to work with me on this to understand how you got a display?
My email is rlust_at_mac.com

Jason2866, 2018-04-30:
@rlust 
Which type of display do you want OLED or LCD Dot Matrix (how many rows and lines)?
To find is really simple. For OLED search for SSD1306 and for LCD with a I2C interface
For example https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20180430063646&SearchText=Ssd1306
or
https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20180430063755&SearchText=Lcd+i2c

rlust, 2018-04-30:
I have an OLED .96 like option 1
128x64
What changes did you make in 06_display.ino?
I am able to scan I2C and see the address 0x3c.

Jason2866, 2018-04-30:
@arendst  Nice, reliable and quick solution. I like it too. In my very early years it was part of my job. Would be nice to own a electric wire rapp machine for private use...

arendst, 2018-04-30:
In my early days, fixing backplanes,  I used a battery powered wire-wrap tool too but the battery has left this planet so I have to use an old, once bought from Tandy wire-wrap manual tool which does the job perfectly.

Jason2866, 2018-04-30:
@rlust  just added in user_config.h
around line 264:
    #define USE_DISPLAY                            // Add I2C code for Diplays 

Nothing changed in 06_display.ino
Time and Date is on the display at start. I´m not at home so i can try the commands from Theo
later the day....

Jason2866, 2018-04-30:
@arendst  I know this hand wrap tool too. Mine still works pretty well after many years! 
And it is a good training for the hand muscels ;-)

rlust, 2018-04-30:
Got it working!!!
It is picking up all of my temperature readings.
This is fantastic! 
Thanks Jason for the guidance!
arendst you made my day, this really is a great feature!!!!!!
Now to pull out my old wire wrap toll and get a larger display.
I also spent to many days back in tech school doing wire wrap :)

arendst, 2018-04-30:
I use a 5V/3A power supply feeding the matrix unit. From there I use USB cables to TFT and LCD. The oled is currently fed by a USB to FTDI converter as the onboard USB connector broke off the board within an hour of use ;-(

ascillato, 2018-04-30:
Amazing! Thanks for sharing :+1:

sharpn, 2018-05-02:
Is there a specific build that needs to be uploaded to the device to get this working?

Jason2866, 2018-05-02:
Yes, branch development-display

bollitec, 2018-05-04:
@arendst, many thanks for your picture with wemos and tft display, can you please send a higher resolution or the pin assignments (wemos - tft) ?

Like to test the branch and have some idea with touch screen display from raspi.

Thanks a lot!

abaszczok, 2018-05-09:
Guys... I can not get it running... I setup display mode model etc... I got OLED 128x64 + D1 mini, which device type did you select in config module?? "generic?" .. I put #define USE_DISPLAY.. what else I'm missing/??? I tested this display + D1 on ESPeasy - the depslay is OK.. any hints?

abaszczok, 2018-05-09:
Got it working.... pull up resistors 4,7k - ESPeasy you dont need to use them - they are internally proramable.. 
@arendst - you'r my hero !!! - keep it rolling

Jason2866, 2018-05-26:
If someone doesnt know how to connect the TFT Display (SPI).
Take a look here:
http://microcontrollerkits.blogspot.de/2015/11/esp8266-wifi-tftlcd.html
CS and DC has to be cofigured over web console in module generic. See picture
[image]
I used not GPIO 2 and 5 as in descriptio example. I used GPIO 15 and 16.

vMeph, 2018-07-08:
hi @arendst was trying this comands not working

```
21:04:49 CMD: displaymode1
21:04:49 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
```

Jason2866, 2018-07-09:
Habe you used the correct branch Development-Display?

vMeph, 2018-07-09:
Yes mate

Jason2866, 2018-07-09:
You have enabled user_config_override.h ? In this file is the display part enabled.
Does the connected Display shows anything? Just out of the box it shows time and date.

vMeph, 2018-07-09:
Got it to work mate

Jason2866, 2018-07-09:
What was missing?

vMeph, 2018-07-09:
enable user_config_override.h  :)

vMeph, 2018-07-09:
@jason2866  anyway to keep values info steady, everytime updates info it keeps moving down
wondering if any command exits to keep them updating in the same place

Jason2866, 2018-07-10:
I had this top, If i remember correct there is a command displaycols and displayrows. You have to set the values here of your display

Dees7, 2018-08-26:
Hello.
Is origin/developmet support Display?
I use ea43540 and connect TFT LCD 1.8 (128x64) with config
[image]
but after save SDA and CLK reset
[image]

Display works but in mirror mode and with lines
[image]

MQT: stat/sonoff/RESULT = {"DisplayModel":4}
MQT: stat/sonoff/RESULT = {"DisplayAddress1":113}
MQT: stat/sonoff/RESULT = {"DisplayRows":8}
MQT: stat/sonoff/RESULT = {"DisplayCols1":21}

BMP280 doesn't work with display in this config.
What do I wrong?

arendst, 2018-08-26:
As the interface is SPI you will need to know it uses 5 signal lines. Tasmota currently only supports hardware SPI. This always uses pins 12, 13 and 14 in addition to user selectable SPI CS and SPI DC.

So in your case connect 12 to MISO (if used)
13 to MOSI
14 to CLK
and any of the other gpios to cs and dc

Other gpios can be used for I2C

arendst, 2018-08-26:
Use displayrotate 0 to 3 for correct rotation.  Only rotation 0 supports hardware scroll.

Dees7, 2018-08-27:
Yes after connnecting as @arendst said BME280 works but display steel mirrored and shows a lot of noise.

Jason2866, 2018-08-27:
Maybe you have to use pullup resistors on all data connections. And a breadboard can make troubles too.
Try to connect everything without it. Use short wires. Electronic can be a pain...

andrethomas, 2018-08-27:
@Dees7 Are you sure about the driver for that lcd?

Dees7, 2018-08-27:
@andrethomas yes I'm shure about LCD. I've tested it.
@Jason2866 you have the same LCD TFT 1.8 tested? This works with test sketch well.

Jason2866, 2018-08-27:
I have this display
https://www.aliexpress.com/item/2-8-inch-TFT-LCD-Module-without-Touch-Panel-ILI9341-Drive-IC-240-RGB-320-SPI/32764300681.html?spm=a2g0s.9042311.0.0.27424c4dWb2ZWj

simo26246, 2018-09-10:
Hi all, I've read through this and am still having a little difficulty. I'm trying to connect a display - 

https://www.aliexpress.com/item/0-96-inch-IIC-SPI-Serial-128X64-Yellow-Blue-OLED-Display-Module-I2C-LCD-Screen-Board/32779485021.html?spm=2114.search0604.3.9.5b3758e8dJJGLc&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_10130_10547_10059_10548_10696_100031_10084_10083_10103_451_10618_5016315_452_10307_10131_10132_5015615_10133,searchweb201603_45,ppcSwitch_5&algo_expid=79e93466-11bf-42de-9e2d-5d49db3c7070-1&algo_pvid=79e93466-11bf-42de-9e2d-5d49db3c7070&transAbTest=ae803_2&priceBeautifyAB=0

But I can't figure out which pins to connect to what, and how to set it up as a generic chip (I'm using a wemos d1 mini) in the config menu.

I know this is super basic, but I'm super new at this - any help would be really appreciated!

Jason2866, 2018-09-11:
You have to connect:
[image]
and to solder the jumpers R1, R4, R6, R7, R8:
[image]
You have bought not the easiest one for beginning :-)

simo26246, 2018-09-12:
Thanks for the reply. That all sounds like it needs to go in the too hard basket, and I'll just buy new screens, haha.

andrethomas, 2018-09-12:
@simo26246 Its simple enough to solder t he bridges where they are marked on the pc board? Why waste a good lcd

simo26246, 2018-09-13:
When you say solder them, you mean just bridge them together with solder?

What would my pin config for a generic device be in the sonoff firmware? Obviously ground and vcc don't need to be setup, but what about the other pins? 

I just need to connect the SDA and SCL? Like this? - (yes, I know this isn't the development branch firmware - it's on another device, and wanted to make sure I got it right before stuffing around again)
[image]

Then connect them to the corresponding pins on the d1 mini? Really appreciate the help guys!

ascillato2, 2018-09-21:
Closing issue as now it is implemented in Tasmota. Thanks everyone for sharing ideas.

closed 2018-09-21 by ascillato2

AalianKhan, 2018-10-27:
is it possible to show the status of a relay on these displays? it will be a really cool feature.

Jason2866, 2018-10-28:
@AalianKhan 
Yes, i agree. But isnt easy. How todo? So many different displays. And if done everyone has different wishes...
You can do it via rules!

last0tt0man, 2018-11-05:
is it  possible to use SSD1306 128x32 model ? I'm trying all combinations but fonts wrong. 
only displaymode 1 date&time working correctly.
[image]
[image]

Jason2866, 2018-11-05:
Try a different font size. If that doesnt help old settings or fragments are in flash
Delete whole flash and try again. I had the same problem...
After a clean start it worked

ceraz68, 2018-11-10:
@simo26246. I'm configured exactly like you Wemos D1 mini and oled shield 64×48 pixels. I flashed the sonoff-dislay.bin 6.3.0 (not sonoff.bin - read release notes). Perform I2Cscan and check displayaddress is set to 60. Don't forget to toggle the wemos On.

Only working command seems to be displaymode. Other commands do not work for me such as displaytext. Displayed date does not scale and can't displaytext.

Jason2866, 2018-11-10:
You have to set DisplayMode 0  to use DisplayText commands.
[DisplayText.docx](https://github.com/arendst/Sonoff-Tasmota/files/2568264/DisplayText.docx)

ceraz68, 2018-11-10:
@Jason2866 : thanks that was really dumb, I was convinced I read somewhere that displaymode 0 meant display off.

Still no luck for me on the wemos mini D1 with oled shield 64×48 pixels. Displaytext doesn't show text in mode 0. Just a black screen.

16:44:52 CMD: displaymode
16:44:52 MQT: stat/sonoff/RESULT = {"DisplayMode":0}
16:45:00 CMD: displayrows
16:45:00 MQT: stat/sonoff/RESULT = {"DisplayRows":8}
16:45:07 CMD: displaycols
16:45:07 MQT: stat/sonoff/RESULT = {"DisplayCols1":25}
16:45:22 CMD: power
16:45:22 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
16:45:22 MQT: stat/sonoff/POWER = ON
16:45:33 CMD: DisplayText [z]
16:45:33 MQT: stat/sonoff/RESULT = {"DisplayText":"[z]"}
16:45:44 CMD: DisplayText [s1l1c1]Hello how are you?
16:45:44 MQT: stat/sonoff/RESULT = {"DisplayText":"[s1l1c1]Hello how are you?"}
16:45:56 CMD: displaymode 2
16:45:56 MQT: stat/sonoff/RESULT = {"DisplayMode":2}

In displaymode 2, text scrolls but truncated. For the second line, I only see "aymode 2" on the screen. So there must be display mis-alignment somewhere. Tried changing the displayrows and displaycols to match font size but this doesn't improve.

I'm wondering if I need to add more settings to the module or the OLED is incompatible. I only assigned  I2C SDA and SCL like simo26246 above.

Jason2866, 2018-11-11:
Hmm, try to "play" with
sp = set text size (scaling factor 1...4) (only for classic GFX font)
fp = set font (1=12, 2=24,(opt 3=8)) if font==0 the classic GFX font is used
I tried the module only (not installed) and i remember i had the same problem
and i got it working with changing one of these...

Jason2866, 2018-11-15:
@simo26246 
I think i knew what is missing to bring your display to working.
See here https://github.com/arendst/Sonoff-Tasmota/issues/4334#issuecomment-439006127
You have for working with Tasmota (I2C) resistors on Position R1, R4, R6, R7 and R8 coud be left open.
**Urgent you have to connect Pin CS with GND!**

raidolo, 2018-11-16:
@Jason2866  downloaded!
I'm sorry, I've just realized that the SH1106 is an I2C display. Mine is SPI. 
The product is this:
https://www.az-delivery.de/collections/displays/products/1-8-zoll-spi-tft-display?ls=en&cache=false&curr=eur  
And it seems it has a ST7735 controller! 

At the moment is not supported even in the Gemu2015 release.

Thanks again anyway, you pointed me in the right direction!
I'll try to include the support for it by myself.

raidolo, 2018-11-16:
Ok I've implemented the ST7735 adafruit Lib in the latest tasmota code. I had to update also the GFX Library as well
It seems working very good now! 

[image]
[image]

ceraz68, 2018-11-18:
Finally got DisplayText working on Wemos D1 mini OLED shield 64x48. As tasmota documented support is limited to 128x64 I expected some limitations. 

"Displaytext [s1l1c1]Hello how are you?"  as described in the wiki wasn't working for me. I tried different size, font, display models. Then I tried using x, y positions instead of line and column .... bingo! 

Upper left position corresponds to x=35 and y=20. Then I realised that displaytext works from line 3 column 7, so the text was being displayed out of the screen bounds, duh!

Changing the number of rows/lines seems to have no effect. Below are my display settings:

CMD: display
MQT: stat/sonoff/RESULT = {"Display":{"Model":2,"Mode":0,"Dimmer":1,"Size":1,"Font":0,"Rotate":0,"Refresh":2,"Cols":[22,8],"Rows":8}}

shivasiddharth, 2018-11-22:
@raidolo , Can you please share the changes you made ?

raidolo, 2018-11-22:
@shivasiddharth, would be a bit long to write it in a post. Essentially I've cloned the ili9341 driver, which basically has the same controls, I've dropped in the Adafruit ST7735 Library and the latest Adafruit GFX 1.3.2. Then added some functions missing in the ST7735 cpp file, such as scrollTo(). Then I've add a new USE_DISPLAY_ST7735 define. I'm thinking of forking the tasmota sources, but it would be near impossibile for me alone to manage that, and I'm not really sure I've done all the things to support the new driver correctly. I'm still learning how tasmota works and something is not really clear, I'm not a full time developer. What do you think is the best way to share the modifications? (I've also integrated the latest IRRemote libraries because I needed the latest ir_Daikin.h)

shivasiddharth, 2018-11-22:
Can you fork https://github.com/gemu2015/Sonoff-Tasmota and upload your changes ? If its too much to ask, may be then you can just zip the folder and attach the zip here.

AalianKhan, 2018-12-01:
hello, I have a 20x4 LCD display I have set the rows to 4 and columns to 20. everything works fine but I can't seem to displays stuff at row 4. I have tried using x and y coordinates but still no luck. here are my display settings 
`{"Display":{"Model":1,"Mode":0,"Dimmer":1,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols" [20,8],"Rows":5}}`

arendst, 2018-12-05:
This sounds correct https://github.com/arendst/Sonoff-Tasmota/issues/2557#issuecomment-416267793 but this also works:
```
2.2 TFT SPI 240x320 --> Wemos D1 mini
TFT ------   Wemos ------
1 Vcc        3V3
2 Gnd        Gnd
3 CS         D8  GPIO15
4 Reset      Rst
5 DC/RS      D0  GPIO16
6 SDI/MOSI   D7  GPIO13
7 SCK        D5  GPIO14
8 LED        D2  GPIO04 
9 SDO/MISO   D6  GPIO12
```
with following config:
[image]

[22 more replies]