arendst/Tasmota #2559: [2.4.1] Device freeze with sleep on

reloxx13, 2018-04-27:
Heya,

before releasing 2.4.1, i would suggest to take a look into the sleep bug.

If i set sleep != 0, the Device is freezing after a few seconds. If this happens to users which got their devices in difficult or not to reach places, it will cause alot of trouble, because the only way to fix it, is serial flashing.

If i comment out in support.ino:
`  WiFi.setSleepMode(WIFI_LIGHT_SLEEP);  // Allow light sleep during idle times`

everything works as espected, but the power consumption is raising.

This is what i found so far, but i was unable to get it fixed:
https://github.com/esp8266/Arduino/issues/4485
https://github.com/esp8266/Arduino/issues/1381
https://github.com/esp8266/Arduino/issues/4658

arendst, 2018-04-28:
Thnx again.

I will test for version 2.4.1 and if it is used do not execute setSleepMode. This way tasmota keeps running while sleepmode is disabled. If people need sleepmode they can revert to 2.4.0

closed 2018-04-29 by arendst

jeylites, 2018-05-06:
How do you turn OFF sleep in core 2.4.0?

After running 2.4.1  for a day I had to downgrade to 2.4.0 due to connectivity issues. 
After a while of running the Sonoffs,  Homebridge loses connection to it despite having wifi, MQTT broker & websever connectivity.  Which wasn't the case in 2.4.0.

Unlike 2.4.0, I could say in 2.4.1, the websever is responsive.

jeylites, 2018-05-06:
How do you turn OFF sleep in core 2.4.0? Will it work if comment out the following or something else has to be changed?
`// WiFi.setSleepMode(WIFI_LIGHT_SLEEP); // Allow light sleep during idle times`

After running 2.4.1 for a day I had to downgrade to 2.4.0 due to connectivity issues.
After a while of running the Sonoffs, Homebridge loses connection to it despite having wifi, MQTT broker & websever connectivity. Which wasn't the case in 2.4.0.

Unlike 2.4.0, I could say in 2.4.1, the websever is responsive.

alton76, 2018-05-06:
> command sleep 0

Thank you!