arendst/Tasmota #2560: Rule Trigger Analog0 possible?

timtimsson, 2018-04-27:
Hello,
is a Trigger Analog0 possible?
`Rule on Tele-Analog0<>=xxxx do publish xxxxx endon`
`Rule on Analog0<>=xxxx do publish xxxxx endon`
Thanks!

arendst, 2018-04-27:
Give it a try and let us know

timtimsson, 2018-04-27:
tried out, unfortunately not possible...

arendst, 2018-04-28:
Was afraid it wouldn't work. Problem is the JSON message. Two solutions possible, I have to change the JSON message or invent code to detect difference between JSON single/second level message value.

I'll leave it for the moment...

timtimsson, 2018-04-28:
.. i would prefer this:  :)))
`xxx/tele/SENSOR {"Time":"2018-04-28T12:50:11","Switch1":"1","Analog0":{"A0":879},"BME280":{"Temperature":25.0,"Humidity":46.1,"Pressure":1000.4},"TempUnit":"C"}`

> I'll leave it for the moment
--> sure, i understand this!

arendst, 2018-04-28:
And then I couldn't leave it as it adds useful functionality.

Give it a go!

timtimsson, 2018-04-29:
... you are a very crazy guy! only a half day an the code is changed!
it works now perfectly with the new rules...

!Thank you for your fantastic work and the great tasmota!

closed 2018-04-29 by timtimsson