arendst/Tasmota #2561: Sonoff Basic, openhab2 and MQTT V3.1 for Windows

dastrix80, 2018-04-28:
Hi All,

First time user of these products.

Ive got my Openhab2 and MQTT broker working as a service.

Ive flashed Tasmota onto my sonoff basic and can toggle the power to the Sonoff via the web browser. 
Attached is the output of my Sonoff MQTT information 

For some reason, I cannot toggle the relay via openHab2 site map. Ive created the items and sitemap and I can see the switch but I cannot triggle it. Commands to the sonoff via the console work fine. Please help :(

**default.items:**
Switch GarageAVReceiver "Garage AV Receiver" <light> (LR,gLight)
    { mqtt=">[192.168.0.3:cmnd/Garage/POWER toggle],
            <[192.168.0.3:stat/Garage/POWER:state:default]" }

**sitemap:**
    	Switch item=GarageAVReceiver        label="Garage AV Receiver"         icon="switch"

**Output from Sonoff Tasmota console:**

02:36:56 CMD: /cmnd/sonoff/power toggle
02:36:56 MQT: stat/Garage/RESULT = {"POWER":"ON"}
02:36:56 MQT: stat/Garage/POWER = ON
02:37:01 MQT: stat/Garage/RESULT = {"POWER":"OFF"}
02:37:01 MQT: stat/Garage/POWER = OFF
02:38:03 MQT: tele/Garage/STATE = {"Time":"2018-04-28T02:38:03", "Uptime":1, "Vcc":3.162, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"Tard", "RSSI":100, "APMac":"28:CF:DA:B7:E0:99"}}

netstat from Broker/OpenHab2 machine (192.168.0.3)

C:\Users\sydney desktop PC>netstat

Active Connections

  Proto  Local Address          Foreign Address        State
  TCP    192.168.0.3:1883       KK:50819               ESTABLISHED
  TCP    192.168.0.3:1883       KK:51246               ESTABLISHED
  **TCP    192.168.0.3:1883       192.168.0.240:3865     ESTABLISHED**  (THIS IS THE SONOFF)
  TCP    192.168.0.3:3389       KK:64747               ESTABLISHED

Frogmore42, 2018-04-28:
It looks like your configuration on Tasmota is confused:
02:36:56 CMD: /cmnd/sonoff/power toggle
 02:36:56 MQT: stat/Garage/RESULT = {"POWER":"ON"}

The CMD says sonoff but the result says Garage.

You are also using a quite old version of Tasmota (5.8.0) and there have been quite a few improvements since then.

I would start with a never version of FM and then completely clear the configuration (with a Reset 2) and reconfigure with the settings you need.

dastrix80, 2018-04-28:
I tried an OTA firmware upgrade to the latest version and had no success, image wouldnt fit even using the minimal version.

Is there a successful way to take it to the latest without the grief?

Jason2866, 2018-04-28:
Follow exact the example in the wiki for OpenHab
https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB
I use OpenHab too. It works perfect and flawessly. At the beginnig OpenHab 
is a mountain to climb. If you learned the basic things it is easy, logic and very powerful
I like it!
You can ask for help here 
https://community.openhab.org

dastrix80, 2018-04-28:
I wish that was the case Jason. Every 'wiki' has it configured differently making it almost impossible unless you're a pgrommer or tech guru. I am not.  Open hab and openhab2 differ. There's no info to suggest any openhab2 Mqtt configuration required.

Jason2866, 2018-04-28:
Post your complete 
Sitemap
Items
and openhab.cfg file
I take a look

dastrix80, 2018-04-28:
> I need time for that too. It is morning here. I have to do other things
> first...

No trouble. Thanks for the email. Shall send you over those details a
little later.

Cheers kris

Jason2866, 2018-04-28:
openhab.cfg ist not used anymore...
Upload mqtt.cfg 
Here is my working one (example)
`#
# Define your MQTT broker connections here for use in the MQTT Binding or MQTT
# Persistence bundles. Replace <broker> with an ID you choose.
#

# URL to the MQTT broker, e.g. tcp://localhost:1883 or ssl://localhost:8883
broker.url=tcp://localhost:1883

# Optional. Client id (max 23 chars) to use when connecting to the broker.
# If not provided a random default is generated.
#<broker>.clientId=<clientId>

# Optional. True or false. If set to true, allows the use of clientId values
# up to 65535 characters long. Defaults to false.
# NOTE: clientId values longer than 23 characters may not be supported by all
# MQTT servers. Check the server documentation.
#<broker>.allowLongerClientIds=false

# Optional. User id to authenticate with the broker.
#<broker>.user=<user>

# Optional. Password to authenticate with the broker.
#<broker>.pwd=<password>

# Optional. Set the quality of service level for sending messages to this broker.
# Possible values are 0 (Deliver at most once),1 (Deliver at least once) or 2
# (Deliver exactly once). Defaults to 0.
#<broker>.qos=<qos>

# Optional. True or false. Defines if the broker should retain the messages sent to
# it. Defaults to false.
#<broker>.retain=<retain>

# Optional. True or false. Defines if messages are published asynchronously or
# synchronously. Defaults to true.
#<broker>.async=<async>

# Optional. Defines the last will and testament that is sent when this client goes offline
# Format: topic:message:qos:retained <br/>
#<broker>.lwt=<last will definition>
`

dastrix80, 2018-04-28:
Hi Jason

here is my sitemap:

sitemap doorbird label="Doorbird sitemap" {
Video url="http://ghbrcn0001:G3QmVcvThM@192.168.0.141/bha-api/video.cgi"
encoding="mjpeg"
Switch item=DoorBird_DoorLight label="Front door light" icon="light"
mappings=[ON="ON"]
Switch item=DoorBird_DoorOpener label="Front door lock" icon="frontdoor"
mappings=[ON="ON"]

Switch item=DoorBird_MotionSensor_Register label="DoorBird Door Motion
Sensor Register" icon="motion"
Switch item=DoorBird_DoorBell_Register label="DoorBird Door Bell Register"
icon="frontdoor"
Switch item=DoorBird_DoorOpen_Register label="DoorBird Door Open Register"
icon="frontdoor"
    Switch item=GarageAVReceiver        label="Garage AV Receiver"
 icon="light"
}

here is my items:

Switch DoorBird_DoorLight "DoorBird Door Light" (DoorBird)  {
http=">[ON:GET:
http://ghbrcn0001:G3QmVcvThM@192.168.0.141/bha-api/light-on.cgi]" }
Switch DoorBird_DoorOpener "DoorBird Door Opener" (DoorBird)  {
http=">[ON:GET:
http://ghbrcn0001:G3QmVcvThM@192.168.0.141/bha-api/open-door.cgi?r=1]" }

Switch DoorBird_MotionSensor "DoorBird Motion Sensor Triggered" (DoorBird)
Switch DoorBird_DoorBell "DoorBird Door Bell Triggered" (DoorBird)
Switch DoorBird_DoorOpen "DoorBird Door Opener Triggered" (DoorBird)

Switch DoorBird_DoorOpen_Register "DoorBird Door Open Register" (DoorBird)
Switch DoorBird_DoorBell_Register "DoorBird Door Bell Register" (DoorBird)
Switch DoorBird_MotionSensor_Register "DoorBird Door Motion Sensor
Register" (DoorBird)

Switch GarageAVReceiver "Garage AV Receiver" <light> (LR,gLight)
    { mqtt=">[192.168.0.3:cmnd/sonoff/garage/POWER:command:*:default],
            <[192.168.0.3:stat/tasmota/garage/POWER:state:default]" }

here is mqtt.cfg

#
# Define your MQTT broker connections here for use in the MQTT Binding or
MQTT
# Persistence bundles. Replace <broker> with an ID you choose.
#

# URL to the MQTT broker, e.g. tcp://localhost:1883 or ssl://localhost:8883
mqtt:cctv.url=tcp://192.168.0.3:1883

# Optional. Client id (max 23 chars) to use when connecting to the broker.
# If not provided a random default is generated.
#<broker>.clientId=<clientId>

# Optional. True or false. If set to true, allows the use of clientId values
# up to 65535 characters long. Defaults to false.
# NOTE: clientId values longer than 23 characters may not be supported by
all
# MQTT servers. Check the server documentation.
#<broker>.allowLongerClientIds=false

# Optional. User id to authenticate with the broker.
#<broker>.user=<user>

# Optional. Password to authenticate with the broker.
#<broker>.pwd=<password>

# Optional. Set the quality of service level for sending messages to this
broker.
# Possible values are 0 (Deliver at most once),1 (Deliver at least once) or
2
# (Deliver exactly once). Defaults to 0.
#<broker>.qos=<qos>

# Optional. True or false. Defines if the broker should retain the messages
sent to
# it. Defaults to false.
#<broker>.retain=<retain>

# Optional. True or false. Defines if messages are published asynchronously
or
# synchronously. Defaults to true.
#<broker>.async=<async>

# Optional. Defines the last will and testament that is sent when this
client goes offline
# Format: topic:message:qos:retained <br/>
#<broker>.lwt=<last will definition>

Jason2866, 2018-04-28:
I am not sure with your broker definition. It looks strange to me
I simply use the name "broker" with this definition
broker.url=tcp://localhost:1883
and with that mqtt commands for a tasmota switch in items look like that
Switch Sonoff_Basic_1  "Couch" <light> (gLight)
    { mqtt=">[broker:cmnd/sonoff-065CA6/power:command:*:default], <[broker:stat/sonoff-065CA6/POWER:state:default]" }

Jason2866, 2018-04-28:
Your items definition is wrong!
Switch GarageAVReceiver "Garage AV Receiver" <light> (LR,gLight)
    { mqtt=">[192.168.0.3:cmnd/sonoff/garage/POWER:command:*:default],
            <[192.168.0.3:stat/tasmota/garage/POWER:state:default]" }

Jason2866, 2018-04-28:
Keep things simple. Use the topics definition from @arendst in the wiki
With that it works. If you understand fully  what you are doing you can change things like you want
But i see no reason for doing that...

dastrix80, 2018-04-28:
> Switch GarageAVReceiver "Garage AV Receiver" (LR,gLight)

So, your sonoff name under MQTT configuration is 'sonoff-065CA6' right?

Like the attached?

I updated my mqtt.cfg file

dastrix80, 2018-04-29:
OK, ive made the changes you've suggested. No joy.

I suspect its a Mosquitto issue talking to openhab

dastrix80, 2018-04-29:
Checking my log in openHab, i see this:

        at java.lang.Thread.run(Unknown Source) [?:?]
19:00:33.645 [ERROR] [.ui.internal.items.ItemUIRegistryImpl] - Cannot retrieve i                                                                                                tem 'GarageAVReceiver' for widget org.eclipse.smarthome.model.sitemap.Switch
19:00:33.645 [ERROR] [.ui.internal.items.ItemUIRegistryImpl] - Cannot retrieve i                                                                                                tem for widget org.eclipse.smarthome.model.sitemap.Switch
19:00:33.646 [ERROR] [.ui.internal.items.ItemUIRegistryImpl] - Cannot retrieve i                                                                                                tem 'GarageAVReceiver' for widget org.eclipse.smarthome.model.sitemap.Switch
19:00:33.646 [ERROR] [.ui.internal.items.ItemUIRegistryImpl] - Cannot retrieve i                                                                                                tem 'GarageAVReceiver' for widget org.eclipse.smarthome.model.sitemap.Switch
19:00:33.646 [ERROR] [.ui.internal.items.ItemUIRegistryImpl] - Cannot retrieve i                                                                                                tem 'GarageAVReceiver' for widget org.eclipse.smarthome.model.sitemap.Switch

dastrix80, 2018-04-29:
Success!!

I uninstalled the MQTT Action and things started to work :D

Jason2866, 2018-04-29:
Strange mqtt action should not disturb. In fact you probably need it later for publishing mqtt commands in OpenHab rules. But congrats to get things working. Maybe Windows isnt the perfect plattform for
mqtt and OpenHab. I am a windows user too, but for this i took the "adventure" Linux.
I was surprised, how fast and easy i got the initial OpenHab running on a Pi3.
To understand and get things running like i wanted is a other thing🤔

dastrix80, 2018-04-29:
There are a few threads on the community with MQTT Action issues ! Odd. Anyway, only using a Sonoff Pro R2 now

closed 2018-05-02 by dastrix80