arendst/Tasmota #2562: Cannot connect via Wifi extender

GileanEU, 2018-04-28:
I have Sonoff TH-16 and I am using Tasmota 5.12 on it. For some reason Sonoff cannot connect via my Wifi extender, but straight to my router's wifi network.

I am going to position my Sonoff in such position that my main wifi network signal is weak, so via extender, I would get stronger signal for Sonoff.

When I enter both my routers wifi and extenders wifi via web UI, Sonoff always connects to my router. If I remove my routers network from Tasmota wifi settings, and leave only extenders network, I cannot connect to Sonoff anymore. My extender is not faulty, as I can connect to extender with my laptop and internet works fine. SSID and passwords are correct as well.

What could cause Sonoff/Tasmota to reject my extenders wifi network?

Jason2866, 2018-04-28:
Is the SSID the same? If not you never get a reliable setup.
Sending the same SSID  is the usual setup in a environment with more accesspoints

GileanEU, 2018-04-28:
But surely I cannot set the same SSID for my extender as my router's wifi network? For example, my main wifi network's SSID is "Mynetwork" and my extender's SSID is "MynetworkEXT". I can connect to both networks from other devices but not with Sonoffs. 

I knew about Tasmota trying one and then another access point, and that's why I tested it so that I removed my routers access point, leaving only my extenders access point, and saved. After that, I could not connect to Tasmota web UI. When I resetted and added my router's access point again via 192.168.4.1, I could connect to Sonoff again.

As for testing which wifi signal is better, I thought that's what Tasmota's wifi-setup window shows when you choose "Scan for wifi networks".

arendst, 2018-04-28:
Provide output of command ``status 0``

GileanEU, 2018-04-28:
In fact, I am not sure if I can or cannot use the same SSID :) In all guides to set up that extender they all changed extender's SSID's to something different. Also, I have both 2,4GHz and 5,8GHz wifi networks provided by my router, and my Comfast repeater supports them both, so when I connect my extender to one of those  (I originally choose 5,8GHz but later changed that to 2,4GHz just to be on the safe side), then I have to type SSID's for both of those two networks. Do you suggest I for example rename my extender's 2,4GHz SSID to same as my router's 2,4GHz SSID?

Here's output of command status 0:
```
11:43:57 CMD: status 0
11:43:57 MQT: stat/sonoffTH2/STATUS = {"Status":{"Module":4,"FriendlyName":"SonoffTH2","Topic":"sonoffTH2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:43:57 MQT: stat/sonoffTH2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:01:47","Sleep":0,"BootCount":10,"SaveCount":27,"SaveAddress":"F9000"}}
11:43:57 MQT: stat/sonoffTH2/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-04-13T06:32:24","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
11:43:57 MQT: stat/sonoffTH2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"JASbel","SSId2":"JASbelEXT","TelePeriod":60,"SetOption":"00000009"}}
11:43:57 MQT: stat/sonoffTH2/STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":504,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
11:43:57 MQT: stat/sonoffTH2/STATUS5 = {"StatusNET":{"Hostname":"sonoffTH2-6048","IPAddress":"192.168.0.105","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:67:37:A0","Webserver":2,"WifiConfig":2}}
11:43:57 MQT: stat/sonoffTH2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.99","MqttPort":1883,"MqttClientMask":"hassiTH-2","MqttClient":"hassiTH-2","MqttUser":"Meitsijas7","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:43:57 MQT: stat/sonoffTH2/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 28 10:43:57 2018","Local":"Sat Apr 28 11:43:57 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
11:43:57 MQT: stat/sonoffTH2/STATUS10 = {"StatusSNS":{"Time":"2018-04-28T11:43:57","DS18B20":{"Temperature":20.8},"TempUnit":"C"}}
11:43:57 MQT: stat/sonoffTH2/STATUS11 = {"StatusSTS":{"Time":"2018-04-28T11:43:57","Uptime":"0T00:01:47","Vcc":3.409,"POWER":"ON","Wifi":{"AP":1,"SSId":"JASbel","RSSI":80,"APMac":"10:FE:ED:2D:A6:8B"}}}
```

arendst, 2018-04-28:
Change wificonfig to 4 and try again.

GileanEU, 2018-04-28:
Done, but nothing happened as far as I can tell?

```
12:00:44 CMD: wificonfig 4
12:00:44 MQT: stat/sonoffTH2/RESULT = {"WifiConfig":"Retry selected"}
12:01:05 CMD: status 0
12:01:05 MQT: stat/sonoffTH2/STATUS = {"Status":{"Module":4,"FriendlyName":"SonoffTH2","Topic":"sonoffTH2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:01:05 MQT: stat/sonoffTH2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:18:54","Sleep":0,"BootCount":10,"SaveCount":28,"SaveAddress":"F8000"}}
12:01:05 MQT: stat/sonoffTH2/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-04-13T06:32:24","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
12:01:05 MQT: stat/sonoffTH2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"JASbel","SSId2":"JASbelEXT","TelePeriod":60,"SetOption":"00000009"}}
12:01:05 MQT: stat/sonoffTH2/STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":504,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
12:01:05 MQT: stat/sonoffTH2/STATUS5 = {"StatusNET":{"Hostname":"sonoffTH2-6048","IPAddress":"192.168.0.105","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"(cencored)","Webserver":2,"WifiConfig":4}}
12:01:05 MQT: stat/sonoffTH2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.99","MqttPort":1883,"MqttClientMask":"hassiTH-2","MqttClient":"hassiTH-2","MqttUser":"Meitsijas7","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:01:05 MQT: stat/sonoffTH2/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 28 11:01:05 2018","Local":"Sat Apr 28 12:01:05 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
12:01:05 MQT: stat/sonoffTH2/STATUS10 = {"StatusSNS":{"Time":"2018-04-28T12:01:05","DS18B20":{"Temperature":20.6},"TempUnit":"C"}}
12:01:05 MQT: stat/sonoffTH2/STATUS11 = {"StatusSTS":{"Time":"2018-04-28T12:01:05","Uptime":"0T00:18:54","Vcc":3.403,"POWER":"ON","Wifi":{"AP":1,"SSId":"JASbel","RSSI":84,"APMac":"10:FE:ED:2D:A6:8B"}}}
12:01:25 MQT: tele/sonoffTH2/STATE = {"Time":"2018-04-28T12:01:25","Uptime":"0T00:19:15","Vcc":3.405,"POWER":"ON","Wifi":{"AP":1,"SSId":"JASbel","RSSI":80,"APMac":"10:FE:ED:2D:A6:8B"}}
```

arendst, 2018-04-28:
Now you van test AP connection by switching to the other AP using command ``ap 0``. Read wiki commands about wificonfig, ap, ssid and password.

GileanEU, 2018-04-28:
Did that, console showed that it changed to my extenders access point, and then...nothing. I cannot connect to it anymore. When this happens, the only thing I can do is restore those working wifi settings manually via 192.168.4.1. And this is what I don't understand, as that other AP, username and password works just fine when I connect to that using my laptop, phone etc...

arendst, 2018-04-28:
To get more info you will have to connect using serial and see what happened to the IP parameters. What IP address did it receive from the WifiExtender if any.

GileanEU, 2018-04-28:
Sorry, I am not sure how to send commands using serial :/ I have connected my programmer to Sonoff and thought I could just send commands with Arduino IDE serial window as there is this "send" button but apparently that doesn't do anything when I try to send "status 0". However, the same serial window does show that Sonoff sends MQTT commands normally and is connected to my extender, or at least it's configured to do so...
```
12:50:54 MQT: tele/sonoffTH2/STATE = {"Time":"2018-04-28T12:50:54","Uptime":"0T00:05:16","Vcc":3.334,"POWER":"ON","Wifi":{"AP":2,"SSId":"JASbelEXT","RSSI":52,"APMac":"40:A5:EF:32:54:F7"}}
12:50:54 MQT: tele/sonoffTH2/SENSOR = {"Time":"2018-04-28T12:50:54","DS18B20":{"Temperature":23.1},"TempUnit":"C"}
```

arendst, 2018-04-28:
Configure your IDE to transmit Both NL & CR at the bottom right in the Serial command window.

GileanEU, 2018-04-28:
Thank you a lot for help!
All right, now I have been tinkering with my router, extender and Sonoff and I came to a conclusion that this extender is a piece of garbage although it does extend wifi signal :) When I managed to send those commands to Sonoff via serial, I noticed that Sonoff got IP address from extender's DHCP server (by default, that was 192.168.10.100). Thing is, that my extender refuses to use the same subnet as my router: My router is 192.168.0.1, and there I have DHCP that gives addresses between 192.168.0.100-199. So I thought to set my extender as static IP 192.168.0.200 (you HAVE to give IP, there is no option to get those from router's DHCP server) and I set gateway as my router's IP address, and gave extender's DHCP server address range of 192.168.0.201-220.

Weird thing is that even when those setting stay in configuration just as I have entered, the real addresses change to 192.168.**1**.200 etc. If I set my router to use IP address 192.168.1.1, then my extender changes to 192.168.0.1, ignoring my settings... And if I disable DHCP, I cannot connect to internet anymore using that extender, because it does not get IP addresses from my router.

So, in the end my Sonoff is now connected to my extender and gets IP address from extender's DHCP server, and sends messages to Home Assistant with MQTT. BUT then I cannot see the IP address of my Sonoff via router's DHCP table and I just have to guess which IP address my repeater has given to it. So, in this case, when I enter 192.168.1.202 in browser, I can access Tasmota configuration page just fine.

I guess I have to live with that, at least now I know my Sonoff uses my extender's stronger wifi signal. :)

Jason2866, 2018-04-28:
A hard way! You did a good job to find this out. You are right the extender is garbage.
No possibility to switch of DHCP server, or when off device is doing no routing what a bad design....

arendst, 2018-04-29:
You can always retrieve the IP address using MQTT commands like ``status 0`` as for MQTT you do not need to know it's IP address. The big bonus of MQTT.

GileanEU, 2018-04-29:
That's good to know, but as I am using Hass.io version of Home Assistant (and Mosquitto broker plugin), I don't think there's any way to send MQTT commands via UI. Would be handy though!

closed 2018-06-28 by stale[bot]