arendst/Tasmota #2563: Failed to learn RfKey on the bridge

DenisAntoine, 2018-04-28:
Hello,

DenisAntoine, 2018-04-28:
I've installed the firmware on a Sonoff RF Bridge and trying to send commands with the console. I'm trying to pair a RF device by using the RfKey1 2 command but it is not recognized. I received a command Unknown response. Do I miss something in the configuration? 
[image]

ascillato, 2018-04-28:
Hi,

On the configuration menu, go to module configuration, and select sonoff bridge.

The command is unknown because you have the sonoff basic module selected

DenisAntoine, 2018-04-29:
Hi,
It was easy. Actually I was lookig for more complex options in the user settings.
Thanks a lot for your support

closed 2018-04-29 by DenisAntoine