arendst/Tasmota #2564: Can't change IP address

dariomartino, 2018-04-28:
Hi guys,

I'm flashing 4 sonoff basic. The first one was successful even if I had some issues with the USB to TTL, but i managed to solve it. I gave it IP 192.168.1.192

When I tried to flash the second one, everything looked fine, but when I checked, it had the same IP (192) and not the one i defined in user_config (193)

I tried restarting atom, rebooting, everything i could possibly think of, but i was unable to flash it with IP 192.168.1.193, it always flashed with IP 192.168.1.192, like the first i flashed.

I'm using a macbook pro (high sierra) with atom/platformio and an arduino mega (the only one I had) to upload.

I'll be very grateful if you can help me figure where the problem is.

Jason2866, 2018-04-28:
Are your sure that you have compiled the firmware with ip xxx.193 ?
You have to clean and than build and flash this firmware.
If you accidently flashed this wrong one time this settings are stored !
You have to reset. This can be done with press and holding the button on the sonoff >4 seconds
Now you can flash your firmware...

Frogmore42, 2018-04-28:
First of all, why are you defining IP addresses instead of using DHCP.  Managing static IP addresses is hard and not necessary for most scenarios.  One of the nice things about Tasmota is that it stores its configuration values outside of the program space, which means you can update the firmware without needing to reconfigure everything and you don't need to manage several copies of the FW each with different IP address settings.

Since you have a mac, you probably already have support for Bonjour which means you can talk to your Tasmota devices by name using HTTP.  If you are using MQTT, and you probably should be, then you don't really need to know the IP address (except when you want to update the FW using the web interface).

I have more than 10 devices running Tasmota at home, none of them have static IP addresses.  My router (an ASUS) does give them the same IP address every time, as do some/many others.  If my router didn't do that, I might be tempted to use static IP addresses.  But, since assigning the same IP address to the same MAC address has been available in routers for about 20 years (my MN-500 had this) I would probably just purchase a new router that had this capability.

ascillato, 2018-06-04:
Hi,

If your issue is solved, please close it. Thanks :+1:

dariomartino, 2018-06-05:
Thanks for the help!

closed 2018-06-05 by dariomartino