arendst/Tasmota #2566: Provide MQTT controllable variables for use in rules

Stevehans, 2018-04-28:
I'd like to be able to push values from MQTT to affect rules (specifically I'd like to push a value for use in a thermostat).

I can see that variables have now been added, it would be great to be able to set these using MQTT commands so the value of a trigger point could be pushed to the device.

Such that the rule could be coded:-

on SENSOR#Temperature>%var1% do.....

arendst, 2018-04-29:
To use values you simply send the complete rule again with your changed value. It's available now.

Stevehans, 2018-04-29:
That would mean duplicating the rules for each of the potential controls that could change a value (and in fact, difficult to change more than one variable).

I'll cover this when I fork and see if you can accept the changes

Stevehans, 2018-05-21:
Custom variables var and mem now addresses this.

Thanks, closing

closed 2018-05-21 by Stevehans