arendst/Tasmota #2567: Is there a way to immediately publish telemetry?

Stevehans, 2018-04-28:
I've had a quick look and it seems that the telemetry is only published when the tele_period expires, is there a way to force an immediate (or almost immediate) refresh of telemetry?

A command to set tele_period to Settings.tele_period -1 I think?

ascillato, 2018-04-29:
Hi,

You can use **rule** as explained on the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules)

Stevehans, 2018-04-29:
Not sure how unless I've missed something, the only call to publish the telemetry is when tele_period is incremented and equals Settings.tele_period?

How would this be done other than by doing individual publish commands to equate to the telemetry output?

I think there needs to be a code change to implement telemetry_refresh that would set the tele_period to Settings.tele_period -1 which would then cause all the actions to occur.

arendst, 2018-04-29:
Use ``status 8`` to retrieve the telemetry data at command.

Stevehans, 2018-04-29:
I'm more interested in the esp publishing it early rather than just viewing it.

I'll fork and make a change later.

Stevehans, 2018-04-29:
The background to this is that I want to provide a refresh button on an mqtt dashboard that will update the telemetry values

Jason2866, 2018-04-29:
I dont understand why status 8 is not the solution for you.
You fire this command with your refresh button and you get immediatly the actual values of all sensors
as a mqtt (Json encoded) Status 8 result

Stevehans, 2018-04-29:
Because my widgets are subscribed to tele/sonoff/sensor

Stevehans, 2018-04-29:
I'm going to re-read the mqtt docs in case I've missed something really basic!

ascillato, 2018-04-29:
Hi,

To do manual updates you use the mqtt broker configuration as explained on https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

Stevehans, 2018-05-01:
Ok, I can see what you're saying, that's fine for say OpenHAB, but the android dashboard takes one subscribe per widget.

I'll fork and add the code I need to achieve this then do a PR.

closed 2018-06-30 by stale[bot]

ascillato, 2018-12-24:
Hi, already fixed. Please update to latest Tasmota. Thanks

markross42, 2019-11-16:
Hi,
 I have the same sort of requirement. And I could not find any solution.

Please, could you send me any information how telemetry data is being sent, after power toggle? No status 8.
(using: tasmota 6.7.1)

Best regards

Mark

meingraham, 2019-11-16:
If a shorter TelePeriod does not meet your need, Rules as posted on Google Groups.

Otherwise modifications to the code by forking the repository yourself.

ma-zal, 2019-12-27:
@ascillato: Looks still not fixed. The command `STATUS 8` is returning incorrectly into mqtt topic `.../STATUS8`. But we are expecting topic `.../SENSOR` to correct understanding by Home Assistant.

**Workaround:**
If you set `TelePeriod` (you can update to same value as current), then Tasmota sends `.../SENSOR` immediatelly. So command `TelePeriod 300` will response all sensor data.

ascillato, 2019-12-27:
I don't know how exactly is your setup but seems for your request that you are not using Tasmota as was designed. Please, address this to the Tasmota support chat to discuss your setup and how to implement it

sep-don-barrett, 2020-01-30:
Here's the use case, which I have too: I need to get rapid updates for certain sensor changes to homeassistant, and have to do a workaround with rules like:
on ENERGY#Current>0.100 do TelePeriod 60 endon

Ideally one would simply use another Status command have the same SENSOR(as opposed to STATE8)  MQTT publishing ordinarily done, to avoid changing various clients down the line, and to have to hardcode the existing teleperiod.  It's the need to push time-critical updates using the usual notification mechanism.  

Seems a very common use case, as push-updates are a common mechanism in control systems.

ascillato, 2020-01-31:
> a special Teleperiod code to force the update without changing the existing teleperiod.

Seems that `teleperiod 1` command is what you are looking for.

sep-don-barrett, 2020-01-31:
teleperiod 1 resets the interval to the firmware default.  I'm suggesting the payload of a rule to force the teleperiod sensor update manually, just as switch operations are pushed at the time of change. The teleperiod reset to force SENSOR broadcast over MQTT I'm using now has to be hardcoded to what I'm otherwise using as the teleperiod of the device, and changed along with that teleperiod should I choose a different one.

ascillato, 2020-01-31:
So, you want a command like the one you are using ( `TelePeriod 60` ), but that will not reset the counter of teleperiod?

sep-don-barrett, 2020-01-31:
Yes, that will push sensor data immediately in the same way as it is normally done, for when conditions are met for which the subscribers have an immediate need to know.  It could be as simple as having "status 8" send a SENSOR broadcast, or having another status or teleperiod argument that triggers the same.

ascillato, 2020-01-31:
Ok. Got it.

Question. The command teleperiod 60 that triggers the tele message and start counting again from 0 up to 60 to send another teleperiod, why is not good that teleperiod time reset in your setup?

An option can be a new argument for teleperiod, like teleperiod 2 to trigger a tele message without resetting the counter. What do you think?

sep-don-barrett, 2020-01-31:
Locally, there's nothing bad about it, it meets my need.  But from good engineering design, it forces the operation to have an unwanted side effect, namely losing the prior value of teleperiod unnecessarily, which has nothing to do with the operation actually sought.  For instance, if I decide that to change the heartbeat teleperiod of a sensor for non-time-critical monitoring, I'd have to change the rule statements for the time-critical pushes, a fragile operation if one is building a larger system based on these concepts, given heartbeat sensor monitoring and time-critical sensor monitoring are somewhat orthogonal in their configuration needs.

By the way, my enormous appreciation for the incredible resource Tasmota has grown into, given the ubiquity of the ESP8266 family in inexpensive consumer devices.

Kedryn, 2020-05-16:
Just came here because i was looking for the same thing. A way to manually force a telemetry message, without changing the telemetry value.

I have, for example, a meteo station outside in the garden, that has several sensors (temp, humidity, pressure ,light). It has also a rain sensor via analog value. My telemetry is set to 10 minutes, but if starts raining, i need read the rain sensor as quick as i can (like 5 seconds)

Right now i have a rule that, if the analog value changes, when is 1024 it sends a teleperiod 600, if it's lower than 1024, it sends a teleperiod 5.

The problem is that if, one day, i need to have a teleperiod of 300, and i set it via de console, as soon as it rains, it will go back to 600. 

If we can have, like a "teleperiod -1" that does not change the actual value but only triggers it, it will be much usefull.

arendst, 2020-05-16:
https://github.com/arendst/Tasmota/issues/2567#issuecomment-385235540

githubuser9999999, 2023-06-20:
I know this is an old thread, but how about just setting the teleperiod as a part of a rule when some threshold is passed and then setting it back when it crosses back. Something like this...

Rule1
  ON ENERGY#Power > 5 DO
    TelePeriod 10
  ENDON

  ON ENERGY#Power < 0 DO
    Teleperiod 300
  ENDON

barbudor, 2023-06-21:
You don't need to specify a period. Using just command `Teleperiod` will trigger a telemetry message as soon as possible
You can also use `Status 10` which is slighly different as it has an additional level in the JSON.

You can also forge your own message and publish it directly through the rule.

Regarding Energy, have you explored all the possibilities of PowerDelta and Margins ?

githubuser9999999, 2023-06-22:
Thanks. I've had success testing the following based on your suggestion. Rule 1 5 sets it to trigger telemetry only once and not repeatedly.

Rule1
  ON ENERGY#Power > 2 DO
    TelePeriod 
  ENDON

  ON ENERGY#Power = 0 DO
    Teleperiod 
  ENDON

rule1 5