arendst/Tasmota #2569: Selecting '43 Sonoff Pow v2' module config no power data displayed on front page of device webserver

dweston, 2018-04-29:
- [x] What hardware you are using
Sonoff POW ver 2.0
Tasmota v 5.12.0m

- [x] Provide the output of command ``status 0``
When module is set to ``43 Sonoff Pow v2``:
``18:42:10 CMD: status 0
18:42:10 MQT: stat/rampLight/STATUS = {"Status":{"Module":43,"FriendlyName":["Ramp Light"],"Topic":"rampLight","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:42:10 MQT: stat/rampLight/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:27","StartupUTC":"2018-04-29T08:41:43","Sleep":0,"BootCount":13,"SaveCount":56,"SaveAddress":"F4000"}}
18:42:10 MQT: stat/rampLight/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-28T17:23:19","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
18:42:10 MQT: stat/rampLight/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxx","xxxxxx"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
18:42:10 MQT: stat/rampLight/STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
18:42:14 MQT: stat/rampLight/STATUS5 = {"StatusNET":{"Hostname":"rampLight-7814","IPAddress":"10.0.1.xx","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":3}}
18:42:14 MQT: stat/rampLight/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7FFE86","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:42:14 MQT: stat/rampLight/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 29 08:42:14 2018","Local":"Sun Apr 29 18:42:14 2018","StartDST":"Sun Oct 07 02:00:00 2018","EndDST":"Sun Apr 01 03:00:00 2018","Timezone":10,"Sunrise":"06:27","Sunset":"17:16"}}
18:42:14 MQT: stat/rampLight/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:42:14 MQT: stat/rampLight/STATUS10 = {"StatusSNS":{"Time":"2018-04-29T18:42:14","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
18:42:14 MQT: stat/rampLight/STATUS11 = {"StatusSTS":{"Time":"2018-04-29T18:42:14","Uptime":"0T00:00:31","Vcc":3.544,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxx","RSSI":92,"APMac":"xx:xx:xx:xx:xx:xx"}}}``

When module is set to ``06 Sonoff Pow``:
[Am unable to extract `status 0` as the device resets immediately the command is issued. The following is the console output following the restart.]]
``00:00:00 Project sonoff Ramp Light (Topic rampLight, Fallback DVES_7FFE86, GroupTopic sonoffs) Version 5.12.0m-2_4_1
00:00:00 WIF: Connecting to AP1 Vista_TC_1 in mode 11N as rampLight-7814...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on rampLight-7814 with IP address 10.0.1.99
18:58:14 MQT: Attempting connection...
18:58:14 MQT: Connected
18:58:14 MQT: tele/rampLight/LWT = Online (retained)
18:58:14 MQT: cmnd/rampLight/POWER = 
18:58:14 MQT: tele/rampLight/INFO1 = {"Module":"Sonoff Pow","Version":"5.12.0m","FallbackTopic":"DVES_7FFE86","GroupTopic":"sonoffs"}
18:58:14 MQT: tele/rampLight/INFO2 = {"WebServerMode":"Admin","Hostname":"rampLight-7814","IPAddress":"10.0.1.xx"}
18:58:14 MQT: tele/rampLight/INFO3 = {"RestartReason":"Software Watchdog"}
18:58:16 MQT: stat/rampLight/RESULT = {"POWER":"ON"}
18:58:16 MQT: stat/rampLight/POWER = ON
18:58:23 MQT: tele/rampLight/STATE = {"Time":"2018-04-29T18:58:23","Uptime":"0T00:00:14","Vcc":3.545,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxx","RSSI":92,"APMac":"xx:xx:xx:xx:xx:xx"}}
18:58:23 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T18:58:23","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"Factor":0.00,"Voltage":239,"Current":0.000}}
18:58:28 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T18:58:28","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":2,"Factor":0.00,"Voltage":240,"Current":0.000}}
18:59:18 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T18:59:18","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"Factor":0.00,"Voltage":240,"Current":0.000}}
18:59:22 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T18:59:22","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":8,"Factor":0.00,"Voltage":240,"Current":0.000}}
18:59:30 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T18:59:30","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":2,"Factor":0.10,"Voltage":240,"Current":0.066}}
18:59:40 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T18:59:40","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"Factor":0.00,"Voltage":240,"Current":0.055}}
18:59:50 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T18:59:50","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":2,"Factor":0.08,"Voltage":240,"Current":0.076}}
19:00:12 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T19:00:12","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"Factor":0.00,"Voltage":240,"Current":0.051}}
19:00:30 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T19:00:30","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":2,"Factor":0.00,"Voltage":239,"Current":0.000}}
19:00:40 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T19:00:40","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"Factor":0.00,"Voltage":240,"Current":0.071}}
``
I've just flashed some brand new Sonoff Pow Version 2.0 modules with v5.12.0.m as I was keen try the timer and rules functions.

During configuration I selected the ``43 Sonoff Pow v2`` module option in the drop down menu as this appeared to best match the module I'm using.

However when wired in place with mains attached I found all the electrical and power measurement displayed on the module front page were all '0' even with relay 'on'.
[Screen Shot 2018-04-29 at 18.43.07.PDF](https://github.com/arendst/Sonoff-Tasmota/files/1958569/Screen.Shot.2018-04-29.at.18.43.07.PDF)

If the config is reset to ``06 Sonoff Pow`` the power data is correctly displayed.
[Screen Shot 2018-04-29 at 18.54.16.PDF](https://github.com/arendst/Sonoff-Tasmota/files/1958572/Screen.Shot.2018-04-29.at.18.54.16.PDF)

The bug appears to fully repeatable.

dweston, 2018-04-29:
Image of board:
[image]

So there are what looks like v2 and real v2 boards then.

arendst, 2018-04-29:
Nope. There is Pow, Pow ver 2.0 and Pow R2

closed 2018-04-29 by arendst

dweston, 2018-04-29:
Ah huh.

Then issue #2570 is a problem.