arendst/Tasmota #2570: v5.12.0m 'status 0' command resets Sonoff POW - {"RestartReason":"Software Watchdog"}

dweston, 2018-04-29:
- [ ] What hardware you are using
Sonoff POW version 2.0
Tasmota v 5.12.0m

- [x] Provide the output of command ``status 0``
``00:00:00 Project sonoff Ramp Light (Topic rampLight, Fallback DVES_7FFE86, GroupTopic sonoffs) Version 5.12.0m-2_4_1
00:00:00 WIF: Connecting to AP1 Vista_TC_1 in mode 11N as rampLight-7814...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on rampLight-7814 with IP address 10.0.1.99
18:58:14 MQT: Attempting connection...
18:58:14 MQT: Connected
18:58:14 MQT: tele/rampLight/LWT = Online (retained)
18:58:14 MQT: cmnd/rampLight/POWER = 
18:58:14 MQT: tele/rampLight/INFO1 = {"Module":"Sonoff Pow","Version":"5.12.0m","FallbackTopic":"DVES_7FFE86","GroupTopic":"sonoffs"}
18:58:14 MQT: tele/rampLight/INFO2 = {"WebServerMode":"Admin","Hostname":"rampLight-7814","IPAddress":"10.0.1.99"}
18:58:14 MQT: tele/rampLight/INFO3 = {"RestartReason":"Software Watchdog"}
18:58:16 MQT: stat/rampLight/RESULT = {"POWER":"ON"}
18:58:16 MQT: stat/rampLight/POWER = ON
18:58:23 MQT: tele/rampLight/STATE = {"Time":"2018-04-29T18:58:23","Uptime":"0T00:00:14","Vcc":3.545,"POWER":"ON","Wifi":{"AP":1,"SSId":"Vista_TC_1","RSSI":92,"APMac":"E8:8D:28:5C:43:25"}}
18:58:23 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T18:58:23","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"Factor":0.00,"Voltage":239,"Current":0.000}}
``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
Am not sufficiently technically adept to know how to create a stack dump.

The Sonoff POW is configured as a `06 Sonoff Pow`` in the config module menu.

If a ``status 0`` command is issued in the device console the device resets itself.

arendst, 2018-04-29:
Tis should work just fine:
```
11:35:16 CMD: status 0
11:35:16 MQT: stat/pow2/STATUS = {"Status":{"Module":6,"FriendlyName":["Pow2"],"Topic":"pow2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:35:16 MQT: stat/pow2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.bin","RestartReason":"Software/System restart","Uptime":"2T02:17:30","StartupUTC":"2018-04-27T07:17:46","Sleep":0,"BootCount":230,"SaveCount":1138,"SaveAddress":"F8000"}}
11:35:16 MQT: stat/pow2/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-26T18:09:29","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
11:35:16 MQT: stat/pow2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"domus1","LogPort":514,"SSId":["indebuurt1","indebuurt2"],"TelePeriod":300,"SetOption":["5580201B","558B8000"]}}
11:35:16 MQT: stat/pow2/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
11:35:16 MQT: stat/pow2/STATUS5 = {"StatusNET":{"Hostname":"pow2","IPAddress":"192.168.2.192","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"5C:CF:7F:A0:16:26","Webserver":2,"WifiConfig":5}}
11:35:16 MQT: stat/pow2/STATUS6 = {"StatusMQT":{"MqttHost":"domus3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A01626","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:35:16 MQT: stat/pow2/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 29 09:35:16 2018","Local":"Sun Apr 29 11:35:16 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
11:35:16 MQT: stat/pow2/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:35:16 MQT: stat/pow2/STATUS10 = {"StatusSNS":{"Time":"2018-04-29T11:35:16","ENERGY":{"Total":56.469,"Yesterday":0.335,"Today":0.158,"Power":13.7,"Factor":0.50,"Voltage":203.8,"Current":0.135}}}
11:35:16 MQT: stat/pow2/STATUS11 = {"StatusSTS":{"Time":"2018-04-29T11:35:16","Uptime":"2T02:17:30","Vcc":3.405,"POWER":"ON","Wifi":{"AP":2,"SSId":"indebuurt2","RSSI":100,"APMac":"00:22:6B:FE:8E:20"}}}
```
If it's reproducible try each status message on it's own to find which one is causing the watchdog.

dweston, 2018-04-29:
``19:42:35 CMD: status 1
19:42:35 MQT: stat/rampLight/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software Watchdog","Uptime":"0T00:44:26","StartupUTC":"2018-04-29T08:58:09","Sleep":0,"BootCount":16,"SaveCount":60,"SaveAddress":"F8000"}}
19:42:43 CMD: status 2
19:42:43 MQT: stat/rampLight/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-28T17:23:19","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
19:43:11 CMD: status 3
19:43:11 MQT: stat/rampLight/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vista_TC_1","dionw"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
19:43:41 CMD: status 4
19:43:41 MQT: stat/rampLight/STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
19:43:55 CMD: status 5
19:43:55 MQT: stat/rampLight/STATUS5 = {"StatusNET":{"Hostname":"rampLight-7814","IPAddress":"10.0.1.99","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"5C:CF:7F:7F:FE:86","Webserver":2,"WifiConfig":3}}
19:44:16 CMD: status 6
19:44:16 MQT: stat/rampLight/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.14","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7FFE86","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:44:34 CMD: status 7
19:44:34 MQT: stat/rampLight/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 29 09:44:34 2018","Local":"Sun Apr 29 19:44:34 2018","StartDST":"Sun Oct 07 02:00:00 2018","EndDST":"Sun Apr 01 03:00:00 2018","Timezone":10,"Sunrise":"06:27","Sunset":"17:16"}}
19:45:05 CMD: status 8
19:45:05 MQT: stat/rampLight/STATUS8 = {"StatusSNS":{"Time":"2018-04-29T19:45:05","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":2,"Factor":0.09,"Voltage":237,"Current":0.078}}}
19:45:12 CMD: status 9
19:45:12 MQT: stat/rampLight/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:45:18 CMD: status 10
19:45:18 MQT: stat/rampLight/STATUS10 = {"StatusSNS":{"Time":"2018-04-29T19:45:18","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":2,"Factor":0.14,"Voltage":237,"Current":0.058}}}
19:45:24 CMD: status 11
19:45:24 MQT: stat/rampLight/STATUS11 = {"StatusSTS":{"Time":"2018-04-29T19:45:24","Uptime":"0T00:47:15","Vcc":3.546,"POWER":"ON","Wifi":{"AP":1,"SSId":"Vista_TC_1","RSSI":90,"APMac":"E8:8D:28:5C:43:25"}}}``

Done individually all ``status`` commands complete.

And trying ``status 0`` again:
``19:49:58 CMD: status 0
19:49:58 MQT: stat/rampLight/STATUS = {"Status":{"Module":6,"FriendlyName":["Ramp Light"],"Topic":"rampLight","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:49:58 MQT: stat/rampLight/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software Watchdog","Uptime":"0T00:51:49","StartupUTC":"2018-04-29T08:58:09","Sleep":0,"BootCount":16,"SaveCount":60,"SaveAddress":"F8000"}}
19:49:58 MQT: stat/rampLight/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-28T17:23:19","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
19:49:58 MQT: stat/rampLight/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vista_TC_1","dionw"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
19:49:58 MQT: stat/rampLight/STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
19:49:58 MQT: stat/rampLight/STATUS5 = {"StatusNET":{"Hostname":"rampLight-7814","IPAddress":"10.0.1.99","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"5C:CF:7F:7F:FE:86","Webserver":2,"WifiConfig":3}}
19:49:58 MQT: stat/rampLight/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.14","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7FFE86","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:49:58 MQT: stat/rampLight/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 29 09:49:58 2018","Local":"Sun Apr 29 19:49:58 2018","StartDST":"Sun Oct 07 02:00:00 2018","EndDST":"Sun Apr 01 03:00:00 2018","Timezone":10,"Sunrise":"06:27","Sunset":"17:16"}}
19:49:58 MQT: stat/rampLight/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:49:58 MQT: stat/rampLight/STATUS10 = {"StatusSNS":{"Time":"2018-04-29T19:49:58","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":3,"Factor":0.28,"Voltage":237,"Current":0.050}}}
19:49:58 MQT: stat/rampLight/STATUS11 = {"StatusSTS":{"Time":"2018-04-29T19:49:58","Uptime":"0T00:51:49","Vcc":3.546,"POWER":"ON","Wifi":{"AP":1,"SSId":"Vista_TC_1","RSSI":90,"APMac":"E8:8D:28:5C:43:25"}}}
19:50:01 MQT: tele/rampLight/SENSOR = {"Time":"2018-04-29T19:50:01","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":1,"Factor":0.08,"Voltage":237,"Current":0.071}}``

Issue not repeatable. I am sorry to waste your time @arendst.

closed 2018-04-29 by arendst