arendst/Tasmota #2571: Time set back to 1970 after OTA update

dweston, 2018-04-29:
- [x] What hardware you are using
- [x] Provide the output of command ``status 0``
``08:00:49 CMD: status 0
08:00:49 MQT: stat/hotWater/STATUS = {"Status":{"Module":6,"FriendlyName":["Hotwater"],"Topic":"hotWater","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
08:00:49 MQT: stat/hotWater/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:48","StartupUTC":"1970-01-01T08:00:01","Sleep":0,"BootCount":44,"SaveCount":190,"SaveAddress":"F7000"}}
08:00:49 MQT: stat/hotWater/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-28T17:23:19","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
08:00:49 MQT: stat/hotWater/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx","xxxxxx"],"TelePeriod":300,"SetOption":["00000009","00000000"]}}
08:00:49 MQT: stat/hotWater/STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
08:00:49 MQT: stat/hotWater/STATUS5 = {"StatusNET":{"Hostname":"hotWater-5879","IPAddress":"10.0.1.xx","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":3}}
08:00:49 MQT: stat/hotWater/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.xx","MqttPort":1883,"MqttClientMask":"SonoffHotWater","MqttClient":"SonoffHotWater","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:00:49 MQT: stat/hotWater/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 08:00:49 1970","Local":"Thu Jan 01 08:00:49 1970","StartDST":"Sun Oct 04 02:00:00 1970","EndDST":"Sun Apr 05 03:00:00 1970","Timezone":10,"Sunrise":"18:47","Sunset":"09:08"}}
08:00:49 MQT: stat/hotWater/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:00:49 MQT: stat/hotWater/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T08:00:49","ENERGY":{"Total":26,"Yesterday":3,"Today":3,"Power":5,"Factor":0.27,"Voltage":244,"Current":0}}}
08:00:49 MQT: stat/hotWater/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T08:00:49","Uptime":"0T00:00:48","Vcc":3.499,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxx","RSSI":16,"APMac":"xx:xx:xx:xx:xx:xx"}}}``

After an OTA firmware update the device time has been set back to 1970. Is there a means of resetting to the present?

Frogmore42, 2018-04-29:
If you have already tried the common things, like restarting, verifying and correcting the NTP server names, what I have found works is to unplug the device completely from power.  Wait about 30min and then plug it back in.  This seems to fix whatever is confused.  My investigation seems to indicate it is below the Tasmota level.  I was hoping it was fixed by a newer set of libraries, but you are using the latest.

The POW devices are not as useful if they don't have the correct time, so pick a time when you can be without it powered.

reloxx13, 2018-04-29:
RSSI: 16, seems a bad connection and the ntp sync failed. 1970 is unix default, means time is not set.

the output is just after 48sec after power on, guess wait more or get a better wifi connection.

also try 2.3.0 framwork, its working better.

dweston, 2018-04-30:
I've tried the 30min removal and reinstatement of power. No change.

The RSSI has improved:
``10:46:08 MQT: stat/hotWater/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T10:46:08","Uptime":"0T02:46:07","Vcc":3.499,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxx","RSSI":20,"APMac":"xx:xx:x:xx:xx:xx"}}}``

Would a further OTA update 'rectify' it?

arendst, 2018-04-30:
Download the latest released version, 5.13.0. It contains a fix for ntp.

dweston, 2018-04-30:
Grt - Thank you.

dweston, 2018-05-01:
I've prepared a .bin file based on v5.13.0 and attempted to OTA upload it to the remote device. The file size is 518,864 bytes. The upload failed with the error displayed: "Upload buffer miscompare".
[Screen Shot 2018-05-01 at 22.47.48.PDF](https://github.com/arendst/Sonoff-Tasmota/files/1963807/Screen.Shot.2018-05-01.at.22.47.48.PDF)
``04:52:55 UPL: File sonoff_5.13.0.bin ...
04:53:05 MQT: Attempting connection...
04:53:05 MQT: Connected
04:53:05 MQT: tele/hotWater/LWT = Online (retained)
04:53:05 MQT: cmnd/hotWater/POWER = 
``
Status report in case this has any bearing:
``05:05:36 CMD: Status 0
05:05:36 MQT: stat/hotWater/STATUS = {"Status":{"Module":6,"FriendlyName":["Hotwater"],"Topic":"hotWater","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
05:05:36 MQT: stat/hotWater/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T21:05:35","StartupUTC":"1970-01-01T08:00:01","Sleep":0,"BootCount":45,"SaveCount":193,"SaveAddress":"FB000"}}
05:05:36 MQT: stat/hotWater/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-28T17:23:19","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
05:05:36 MQT: stat/hotWater/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx","xxxxxx"],"TelePeriod":300,"SetOption":["00000009","00000000"]}}
05:05:36 MQT: stat/hotWater/STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
05:05:36 MQT: stat/hotWater/STATUS5 = {"StatusNET":{"Hostname":"hotWater-5879","IPAddress":"10.0.1.xx","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":3}}
05:05:36 MQT: stat/hotWater/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.14","MqttPort":1883,"MqttClientMask":"SonoffHotWater","MqttClient":"SonoffHotWater","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
05:05:36 MQT: stat/hotWater/STATUS7 = {"StatusTIM":{"UTC":"Fri Jan 02 05:05:36 1970","Local":"Fri Jan 02 05:05:36 1970","StartDST":"Sun Oct 04 02:00:00 1970","EndDST":"Sun Apr 05 03:00:00 1970","Timezone":10,"Sunrise":"18:48","Sunset":"09:09"}}
05:05:36 MQT: stat/hotWater/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
05:05:36 MQT: stat/hotWater/STATUS10 = {"StatusSNS":{"Time":"1970-01-02T05:05:36","ENERGY":{"Total":30,"Yesterday":3,"Today":7,"Power":5,"Factor":0.35,"Voltage":244,"Current":0}}}
05:05:36 MQT: stat/hotWater/STATUS11 = {"StatusSTS":{"Time":"1970-01-02T05:05:36","Uptime":"0T21:05:35","Vcc":3.499,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxx","RSSI":10,"APMac":"xx:xx:xx:xx:xx:xx"}}}``

dweston, 2018-05-01:
OTA upgrade to v5.13.0 successful.

However original issue persists:
``00:00:00 Project hotWater Hotwater (Topic hotWater, Fallback SonoffHotWater, GroupTopic sonoffs) Version 5.13.0-2_4_1
00:00:00 WIF: Connecting to AP1 xxxxxx in mode 11N as hotWater-5879...
00:00:03 RSL: tele/hotWater/SENSOR = {"Time":"1970-01-01T00:00:03","ENERGY":{"Total":23,"Yesterday":3,"Today":0,"Power":16,"Factor":0.45,"Voltage":246,"Current":0}}
00:00:04 RSL: tele/hotWater/SENSOR = {"Time":"1970-01-01T00:00:04","ENERGY":{"Total":23,"Yesterday":3,"Today":0,"Power":6,"Factor":0.17,"Voltage":246,"Current":0}}
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on hotWater-5879.local with IP address 10.0.1.xx
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/hotWater/LWT = Online (retained)
00:00:06 MQT: cmnd/hotWater/POWER = 
00:00:06 MQT: tele/hotWater/INFO1 = {"Module":"Sonoff Pow","Version":"5.13.0","FallbackTopic":"SonoffHotWater","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/hotWater/INFO2 = {"WebServerMode":"Admin","Hostname":"hotWater-5879","IPAddress":"10.0.1.xx"}
00:00:06 MQT: tele/hotWater/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: tele/hotWater/SENSOR = {"Time":"1970-01-01T00:00:06","ENERGY":{"Total":23,"Yesterday":3,"Today":0,"Power":12,"Factor":0.30,"Voltage":246,"Current":0}}
00:00:07 MQT: stat/hotWater/RESULT = {"POWER":"ON"}
00:00:07 MQT: stat/hotWater/POWER = ON
``

arendst, 2018-05-01:
Right. Well first of all the time is not set back; it is just never set and the current time is the time after a reboot without NTP synchronization.

You will have to test your NTP configuration using ``ntpserver`` commands (See wiki for possible options).

Enabling more logging (``seriallog 3`` or ``weblog 3``) might help in finding out what your local problem is.

dweston, 2018-05-02:
Noted.

With ``seriallog 3`` set this is the reboot result:
``00:00:00 CFG: Loaded from flash at FA, Count 214
00:00:00 APP: Boot Count 50
00:00:00 Project hotWater Hotwater (Topic hotWater, Fallback DVES_EA96F7, GroupTopic sonoffs) Version 5.13.0-2_4_1
00:00:00 CFG: Saved to flash at F9, Count 215, Bytes 2560
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 xxxxxx in mode 11N as hotWater...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 RSL: tele/hotWater/SENSOR = {"Time":"1970-01-01T00:00:03","ENERGY":{"Total":25,"Yesterday":3,"Today":2,"Power":7,"Factor":0.20,"Voltage":246,"Current":0}}
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on hotWater.local with IP address 10.0.1.xx
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/hotWater/LWT = Online (retained)
00:00:07 MQT: cmnd/hotWater/POWER = 
00:00:07 MQT: Subscribe to cmnd/hotWater/#
00:00:07 MQT: Subscribe to cmnd/sonoffs/#
00:00:07 MQT: Subscribe to cmnd/DVES_EA96F7/#
00:00:07 MQT: tele/hotWater/INFO1 = {"Module":"Sonoff Pow","Version":"5.13.0","FallbackTopic":"DVES_EA96F7","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/hotWater/INFO2 = {"WebServerMode":"Admin","Hostname":"hotWater","IPAddress":"10.0.1.xx"}
00:00:07 MQT: tele/hotWater/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: tele/hotWater/SENSOR = {"Time":"1970-01-01T00:00:07","ENERGY":{"Total":25,"Yesterday":3,"Today":2,"Power":11,"Factor":0.38,"Voltage":244,"Current":0}}
00:00:08 MQT: stat/hotWater/RESULT = {"POWER":"ON"}
00:00:08 MQT: stat/hotWater/POWER = ON
``
If I review the same output from one of the other Sonoff Pow devices I have working and which have only been flashed once with v5.12.0m:
``00:00:00 Project sonoff Laundry (Topic laundry, Fallback DVES_7FFC34, GroupTopic sonoffs) Version 5.12.0m-2_4_1
00:00:00 WIF: Connecting to AP1 xxxxx in mode 11N as laundry...
17:19:48 WIF: Connected
17:19:48 HTP: Web server active on laundry with IP address 10.0.1.xx
17:19:49 MQT: Attempting connection...
17:19:49 MQT: Connected
17:19:49 MQT: tele/laundry/LWT = Online (retained)
17:19:49 MQT: cmnd/laundry/POWER = 
17:19:49 MQT: tele/laundry/INFO1 = {"Module":"Sonoff Pow","Version":"5.12.0m","FallbackTopic":"DVES_7FFC34","GroupTopic":"sonoffs"}
17:19:49 MQT: tele/laundry/INFO2 = {"WebServerMode":"Admin","Hostname":"laundry","IPAddress":"10.0.1.xx"}
17:19:49 MQT: tele/laundry/INFO3 = {"RestartReason":"Software/System restart"}
17:19:50 MQT: stat/laundry/RESULT = {"POWER":"ON"}
17:19:50 MQT: stat/laundry/POWER = ON
``

As all my devices use the same NTP server settings as follows:
``#define NTP_SERVER1            "0.au.pool.ntp.org"
#define NTP_SERVER2            "1.au.pool.ntp.org"
#define NTP_SERVER3            "2.au.pool.ntp.org"
``
As the 2nd Sonoff module is setting time correctly I assume that these settings are correct and are working. And I've checked and these are definitely the same settings I used with the v5.13.0 firmware set-up.

As a triple check I applied the v5.13.0 firmware OTA to a third Sonoff Pow [one that was originally serial flashed once only with v5.12.0m] and that produced an interesting outcome.

After the initial restart this third device came up with the same time setting problem. However after reviewing the device again a couple hours later restarting it, I find the issue on that device at least has cleared itself.
``00:00:00 APP: Set Baudrate to 4800
00:00:00 Project hotWater Ramp Light (Topic rampLight, Fallback DVES_7FFE86, GroupTopic sonoffs) Version 5.13.0-2_4_1
00:00:00 WIF: Connecting to AP1 xxxxxx in mode 11N as ramplight...
17:09:02 WIF: Connected
17:09:02 DNS: Initialized
17:09:02 HTP: Web server active on ramplight.local with IP address 10.0.1.xx
17:09:03 MQT: Attempting connection...
17:09:03 MQT: Connected
17:09:03 MQT: tele/rampLight/LWT = Online (retained)
17:09:03 MQT: cmnd/rampLight/POWER = 
17:09:03 MQT: tele/rampLight/INFO1 = {"Module":"Sonoff Pow","Version":"5.13.0","FallbackTopic":"DVES_7FFE86","GroupTopic":"sonoffs"}
17:09:03 MQT: tele/rampLight/INFO2 = {"WebServerMode":"Admin","Hostname":"ramplight","IPAddress":"10.0.1.xx"}
17:09:03 MQT: tele/rampLight/INFO3 = {"RestartReason":"Software/System restart"}
17:09:05 MQT: stat/rampLight/RESULT = {"POWER":"ON"}
17:09:05 MQT: stat/rampLight/POWER = ON
``
So this issue appears to be limited to the circumstances on only one particular device, which happes to be the device that was the original module I first flashed.

arendst, 2018-05-02:
NTP is synced at startup after around 8 seconds. If it fails it stops. NTP is synced once every hour. The moment of sync is determined by the device MAC address to spread NTP call flooding.

So at some time the NTP sync should succeeded or if it doesn't the comms to the NTP server is unstable.

dweston, 2018-05-02:
@arendst Low RSSI could be an issue in this instance, though I've not recently had any difficulty communicating with the device myself, including several OTA firmware upgrades. The RSSI levels on the device are fluctuating quite significantly at times - see attached image.
[image]

But of more significance is the responses to the ``NtpServer`` command.
The device exhibiting the issues did not have a NTP server set and was not responding to commanding it to accept one.
``NtpServer`` in the console produced:
 ``MQT: stat/hotWater/RESULT = {"NtpServer1":" "}
And subsequent attempts to install one via:
``NtpServer1 1.au.pool.ntp.org``
Did not work.
But persistence appears to have paid off:
Bingo!!
Reissuing ``NtpServer1 1.au.pool.ntp.org`` just now has been accepted and the device has rebooted:
``00:00:00 CFG: Loaded from flash at F6, Count 218
00:00:00 APP: Boot Count 51
00:00:00 Project hotWater Hotwater (Topic hotWater, Fallback DVES_EA96F7, GroupTopic sonoffs) Version 5.13.0-2_4_1
00:00:00 CFG: Saved to flash at F5, Count 219, Bytes 2560
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 xxxxxx in mode 11N as hotWater...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 RSL: tele/hotWater/SENSOR = {"Time":"1970-01-01T00:00:03","ENERGY":{"Total":25,"Yesterday":3,"Today":3,"Power":541,"Factor":0.86,"Voltage":242,"Current":3}}
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on hotWater.local with IP address 10.0.1.xx
00:00:06 APP: (UTC) Wed May 02 14:26:52 2018, (DST) Sun Oct 07 02:00:00 2018, (STD) Sun Apr 01 03:00:00 2018
00:26:53 MQT: Attempting connection...
00:26:53 MQT: Connected
00:26:53 MQT: tele/hotWater/LWT = Online (retained)
00:26:53 MQT: cmnd/hotWater/POWER = 
00:26:53 MQT: Subscribe to cmnd/hotWater/#
00:26:53 MQT: Subscribe to cmnd/sonoffs/#
00:26:53 MQT: Subscribe to cmnd/DVES_EA96F7/#
00:26:53 MQT: tele/hotWater/INFO1 = {"Module":"Sonoff Pow","Version":"5.13.0","FallbackTopic":"DVES_EA96F7","GroupTopic":"sonoffs"}
00:26:53 MQT: tele/hotWater/INFO2 = {"WebServerMode":"Admin","Hostname":"hotWater","IPAddress":"10.0.1.xx"}
00:26:53 MQT: tele/hotWater/INFO3 = {"RestartReason":"Software/System restart"}
00:26:54 MQT: stat/hotWater/RESULT = {"POWER":"ON"}
00:26:54 MQT: stat/hotWater/POWER = ON
00:27:01 MQT: tele/hotWater/STATE = {"Time":"2018-05-03T00:27:01","Uptime":"0T00:00:15","Vcc":3.499,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxx","RSSI":16,"APMac":"xx:xx:xx:xx:xx:xx"}}
00:27:01 MQT: tele/hotWater/SENSOR = {"Time":"2018-05-03T00:27:01","ENERGY":{"Total":25,"Yesterday":3,"Today":3,"Period":2,"Power":539,"Factor":0.87,"Voltage":243,"Current":3}}
``
Now the response to ``NtpServer`` is:
``00:40:24 CMD: NtpServer
00:40:24 RSL: Group 0, Index 1, Command NTPSERVER, Data 
00:40:24 MQT: stat/hotWater/RESULT = {"NtpServer1":"0.au.pool.ntp.org"}
``
Many, many thanks for your perseverance @arendst . Looks like that nailed it.

@Frogmore42 I tried the 30mins removal from power approach 24hrs ago but that did not achieve anything here.

Looks like it was/is a consequence of low RSSI leading to an aberrant response to NTP syncing. Anyhow - looks like using the ``NtpSever`` commands sorted it out.

dweston, 2018-05-03:
I knew it wasn't an NTP server naming issue as all the other modules were working fine using the same server details.

As you say - good its working now. Thanks for your guidance.

Frogmore42, 2018-05-03:
Based on your previous post, it was an NTP server name issue.  The server name got corrupted.  I asked you to verify the device had the correct value.  Some other people have seen some settings get corrupted some times.  That is why it is always good to verify what the devices think to make sure it matches what you want/set in the config file.  Until one understands the details of how the settings get set, it can be confusing that they don't seem to be taking.

But, if you are having significant signal strength issues, you should address that too.

dweston, 2018-05-03:
Ah yes - I understand you more clearly now that I better comprehend the ``NtpServer`` commands and @arendst 's insights into how the NTP code functions.

closed 2018-05-03 by arendst