arendst/Tasmota #2573: IR command send with terminal

prasad7676, 2018-04-29:
How to send irsend command from terminal using curl

i tried.... curl -H "Content-Type: application/json" -X POST  http://192.168.1.19/cm?cmnd=irsend -d '{"Protocol": "NEC","Bits": "32","Data": "551489775"}'

also tried....curl -k "http://192.168.1.19/cm?cmnd=irsend {"Protocol": "NEC","Bits": 32,"Data": 551489775}

get error :......{"IRSend":"No PROTOCOL, BITS or DATA"}

Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
RPI
- [ ] Provide the output of command ``status 0``
17:27:30 MQT: stat/LED/STATUS = {"Status":{"Module":19,"FriendlyName":["Development Board "],"Topic":"LED","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Jason2866, 2018-04-29:
May you look here:
#2457

prasad7676, 2018-04-29:
thanks jason but i want to setup to home assistant so im trying to use command line switch.

where i have to use curl -k "http://192.168.1.19/cm?cmnd=irsend {"Protocol": "NEC","Bits": 32,"Data": 551489775}"

Jason2866, 2018-04-29:
I really dont know, so i ask this question. Has Home Assistant no mqtt support?
Tasmota is powerful and easy with mqtt...

prasad7676, 2018-04-29:
HA support Mqtt but i dont know how to send ir command on mqtt. do you?

prasad7676, 2018-04-29:
thanks man to bring to my attention (MQTT)
i got it but still cant get state

- platform: mqtt
  name: "TV"
  state_topic: "tele/LED/RESULT" 
  command_topic: "cmnd/LED/irsend"
  payload_on: '{"Protocol": "NEC","Bits": 32,"Data": 551489775 }'
  payload_off: '{"Protocol": "NEC","Bits": 32,"Data": 551489775 }'
  optimistic: true
  retain: false
  qos: 0

Jason2866, 2018-04-29:
Sorry i cant help you here. It is from the HA dependent.
You have to parse the JsonPath encoded result string.
You can watch this with a mqtt "spy" Take a look here under integration
https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB

roykaandorp, 2018-04-30:
You could do it like this: curl -G "http://192.168.1.19/cm" --data-urlencode "cmnd=IRsend {"Protocol":"NEC","Bits":32,"Data":551489775}"

prasad7676, 2018-04-30:
is there anyway to get state for command?

closed 2018-06-30 by stale[bot]