arendst/Tasmota #2574: Enable to set option14 only for 2 channels by using a sonoff 4CH pro

smartboart, 2018-04-29:
Hello i am using a sonoff 4CH pro. It would be great to have the possibility to set option 14 only for 2 channels...For example: Channel 1 and 2 have the interlock function activ and channel 3 and 4 can be on or off ...the interlock from channel 1 and 2 have no effect to 3 and 4... 
or channel 3 and 4 have interlock and 1 and 2 not...Or interlock for 1 and 2 and interlock for 3 and 4...
This will make it the 4CH more usefull...
With the dip switches on the 4CHpro it is just possible to set interlock for all channels...even with the command setoption14 1/on..
i would be very happe to have these options , which i described ..Maybe others too...

Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

localhost61, 2018-06-14:
I agree, see #824.

stefanbode, 2018-06-26:
@smartboart If interested see setOption31 in my branch of Tasmota. you will find it under stefanbode. There is also a wiki explaining the functionality and how to use it. Should be what you're looking for.

ascillato2, 2018-08-25:
Duplicated issue. See #824.

closed 2018-08-25 by ascillato2