arendst/Tasmota #2575: Changing user_config.h file TIME_DST and TIME_STD

mxplea10, 2018-04-29:
I live in the United States in the central time zone.  Daylight Saving Time (DST) starts the 2nd week of March on  Sunday at 0200 and ends the 1st week of November on Sunday at 0200.  I modified the user_config.h file to reflect local DST with the following:

```
#define TIME_DST               North, Second, Sun, Mar, 2, +120
#define TIME_STD               North, First, Sun, Nov, 2, +60
```

After the Sonoff Basic is flashed with the modified Tasmota user_config.h file, the Sonoff configuration can not be access.  I had to change the user_config.h file back to the default TIME_DST and TIME_STD configuration and flash the Sonoff again to access the Sonoff's configuration.

```
#define TIME_DST               North, Last, Sun, Mar, 2, +120 
#define TIME_STD               North, Last, Sun, Oct, 3, +60
```

What are the correct configurations for TIME_DST and TIME_STD for my time zone?  What programming language is this and where can I find the commands, syntax and their options for future reference?  

I can manually set the Sonoff to the DST offset (-5) but I would like the time to change automatically.

[wiringlight status 0 Output.txt](https://github.com/arendst/Sonoff-Tasmota/files/1959155/wiringlight.status.0.Output.txt)

curzon01, 2018-04-30:
> What are the correct configurations for TIME_DST and TIME_STD for my time zone?

The conditions are correct but you have to define the hour difference from UTC to CT in minutes.
For CT are -5 h = -360 minutes for STD and -6 h = -360 minutes for DST:
```
#define TIME_DST               North, Second, Sun, Mar, 2, -360
#define TIME_STD               North, First, Sun, Nov, 2, -300

```
> What programming language is this

Thee are C language enums, for possible value see file sonoff.h line 140:
```
enum WeekInMonthOptions {Last, First, Second, Third, Fourth};
enum DayOfTheWeekOptions {Sun=1, Mon, Tue, Wed, Thu, Fri, Sat};
enum MonthNamesOptions {Jan=1, Feb, Mar, Apr, May, Jun, Jul, Aug, Sep, Oct, Nov, Dec};
enum HemisphereOptions {North, South};
```
> where can I find the commands, syntax and their options for future reference?

simply use it in the order of the example devided by commas
`<Hemisphere>, <WeekInMonth>, <DayOfTheWeek>, <MonthNames>, <Hour>, <UTCOffsetInMinutes>`

Be aware when changing values within user_config.h and load the firmware to same device where you previously used other user_config.h values, it will not be used for default.

If you want Tasmota uses your new values from user_config.h the easiest way is to use the command `Reset 1` (see [Maintenance](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management) after flashing commands for details).

mxplea10, 2018-04-30:
Can these commands be defined in the Sonoff Module Console without having to flash each Sonoff again?  If so, how are they set in the Console?

JueBag, 2018-05-01:
What is your "Timezone" setting, it should be 99!

curzon01, 2018-05-01:
You have to set `Timezone 99` to get TIME_* defines active, see [Wiki Timezone](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management).

Currently your local daylight settings needs to set within user_config.h

closed 2018-05-01 by mxplea10