arendst/Tasmota #2576: Issues reading correct temperature from DS18B20 on Sonoff TH

sjev, 2018-04-29:
I have a Sonoff TH (Ver 2.1) with a DS18B20 sensor attached.
The problem is that it often reads 85 C, approx half of the time.  
I suspected the sensor itself, but the original sonoff firmware can read the temperature just fine. (after I've reflashed it back)
... this is most probably not tasmota specific problem, as I was also getting incorrect temperature readings with micropython.
Anybody got a clue what could be causing this?

P.S. AM2301 works just fine, but I need the DS18B20

apastuszak, 2018-04-29:
I am running 5.12.0 on a TH10 and have had zero issues with the DS18B20 temperature probe.  What version of Tasmota are you running?

sjev, 2018-04-29:
...the latest  version that I've pulled from git today.

Frogmore42, 2018-04-30:
The sensor has a default reading of 85C, ie that is the reading you get before it has completed a measurement.  I have seen this when a sensor is not given proper power.  How do you have it connected?
Are you using a breadboard?  I just got a batch that are really bad at making connections.

sjev, 2018-04-30:
I've got a Sonoff TH10. The sensor is connected with 2.5 mm jack. The odd thing is that original Sonoff firmware does not have issues reading the sensor, but both Tasmota and Micropython struggle with it. In fact while Tasmota gets a correct value half of the time, I was not able to get a single correct reading with micropython. I've checked the pullup by the way,  it seems to be correct 5kOhm between input and 3.3V

apastuszak, 2018-04-30:
Have you tried the current stable release of Tasmoto (5.12.0).?  That's
working flawlessly for me.

Frogmore42, 2018-04-30:
I have a TH10 that had the same problem.  I replaced the 2.5mm jack and have not had a problem since.  The quality of those jacks is very poor.  Mine felt very loose.

Frogmore42, 2018-04-30:
The original FW might be smarter about rejecting reads of 85 and retrying.  That was what I did in FW I wrote for a different platform.  0 is also suspect as it is what you get if the data line is held low and it still passes the CRC.  Getting reliable data from sensors is never easy.

sjev, 2018-04-30:
Solved :D . There was indeed a faulty connection, thank you for pointing me in the good direction. :+1: 
It turned out to be the sensor cable where the fault was located.  I've cut off the male jack of the sensor and soldered the wires directly to the board. After testing the jack that came off the sensor, I've discovered that there was an unreliable Vcc connection.

closed 2018-04-30 by sjev