arendst/Tasmota #2579: Dear Community, a Noob needs help

KBlobner, 2018-04-30:
Hallo Community

I have tried to flash my Basic and S20 to TASMOTA, but I wasn't able to do it OTA or to get a working FTDI or a FTDI working. This depends on how you see it.
So I wanted to ask you if someone here is from Munich, Germany or near by and could help me with the flashing.

Thank you and best regards 
Kilian

Jason2866, 2018-04-30:
OTA isnt working anymore if stock firmware is a newer. (i think >1.6x it stopps working)
FTDI devices are know to be faked.... The faked ones are often making troubles.
Give us a little bit more info, we try to help. Have you soldered header pins?
What PC, enviroment you are using? A good Platform is Atom.IO for compiling Tasmota yourself.
For the start a ready compiled one is a good choice!

titomontes, 2018-04-30:
The easiest way is to review every step you have taken. There ase several troubleshooting videos in youtube, you should take a look. 
First, you need to make absolutely sure that every contact is... making contact. Second, make sure your FTDI  board is working properly (you should review the driver and see if it´s properly installed), and that your sonoff is entering uploading mode (pressing the button when power -ie 3,3v contact- is connected), there should be a solid red light.

reloxx13, 2018-04-30:
here is a good german community: http://forum.creationx.de/forum/

otherwise give the requested infos to hep you :)

KBlobner, 2018-04-30:
Thanks guys
I am working with a MacOS 10.11 and have Atom.IO installed. I saw a good youtube video on how to compile Tasmota, but I have huge problems with the FTDI.
 It seems that it isn't working with MacOS
I tried it on different ways but it always seems that AtomIO isn't seeing any FTDI device

reloxx13, 2018-04-30:
Some user notices that the current output of a mac or an usb is not enough to flash a device

RaymondMouthaan, 2018-05-01:
(@reloxx13 ) Just like to inform you I have no issues flashing devices on my iMac (27-inch, Mid 2011) using the FTD1232.

MYLE-01, 2018-05-01:
i have a FT232RL 3.3V 5.5V FTDI USB to TTL Serial Adapter Module 

it has not miss a beat just have to remember to put it into 3.3v when working with SONOFF

RaymondMouthaan, 2018-05-01:
@Frogmore42 do you have a link of the FTDI adapter which works best (no fake clone)?

Frogmore42, 2018-05-01:
None of the ones I have seen have an external onboard regulator.  They all depend on the converter chip to provide the 3V3.  This will work for some things.  To do it right you need to use a regulator to drop the USB bus voltage down.  The real FTDI converter I have has the VCC output pin tied to USB VBUS (5V), despite it being labeled 3V3.  I did not notice this at first and did not see any ill effects, but it is operating the esp8266 outside of its maximum rating.  So, I don't recommend doing that.  It turns out the 3V3 label refers to the logic levels for the serial port.

Since many people are successful in programming their devices I can only assume that there are differences in parts and sometimes it works.  But, I also see many people posting that they're having problems, so it doesn't always work out.

What I do is put a small 3V3 regulator between 5V and the VCC pin on a Sonoff.  I also use a PC that I know can provide enough power or put a powered USB hub in between the PC and the adapter.  If I had a MacBook I would use the powered hub.  Powered hubs are designed to provide more power.  Also use a USB 3 hub port, because they are required to provide more power.

KBlobner, 2018-05-03:
Problem solved
I bought a ch340 chip (this one here https://www.amazon.de/gp/product/B072K3Z3TL/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1)
and it worked right at the first time

Thank you all to explain everything to me

Now I just have to make it work with my Homebridge,
but this is another topic

thanks again

closed 2018-05-03 by KBlobner