arendst/Tasmota #2580: New rule for tele- does not work

roblad, 2018-04-30:
Hi,

could you see to last post

https://github.com/arendst/Sonoff-Tasmota/issues/2517

arendst, 2018-04-30:
Pls provide all requested info when you openend this issue.

roblad, 2018-04-30:
Hi,

 [sonoff 4ch pro] 

Tasmota v5.13.0

I have attached 5 DS18B20 to GPIO2 with the set: #define USE_DS18x20 , USE_RULE defined as well.
I am trying to do such rule:
21:33:37 CMD: rule
21:33:37 MQT: stat/podlewaczka/RESULT = {"Rule":"ON","Once":"OFF","Rules":"on TELE-ds18b20-2#Temperature do publish domoticz/in {"idx":120,"nvalue":0,"svalue":"%value%"} endon"}

the rule does not work

debug does not shows anything.

sensor string

SENSOR = {"Time":"2018-04-30T21:36:31","DS18B20-1":{"Id":"0000053808C1","Temperature":30.50},"DS18B20-2":{"Id":"0000054BDDF6","Temperature":24.50},"DS18B20-3":{"Id":"0000054C3854","Temperature":27.50},"DS18B20-4":{"Id":"0000054CCEC4","Temperature":45.88},"DS18B20-5":{"Id":"0000054D6492","Temperature":23.38},"AM2301":{"Temperature":24.40,"Humidity":48.6},"Gas":{"Type":"CLEANAIR","ppm":0},"TempUnit":"C"}

arendst, 2018-05-01:
The problem is with your SENSOR JSON string length and my buffer size of 400 bytes.

According to the ArduinoJson Assistant my buffer should have a size of 800 bytes to satisfy your need. I did a test and indeed that works and the rule is triggered.

I'll change the buffer size in the next development release to 1000 to satisfy most needs.

roblad, 2018-05-01:
Working , Thanks !

closed 2018-05-01 by arendst

Dylantje, 2018-08-30:
Dear how do you get on the dallas id?

And please a how to for the use of multiple dallas on a sonoff dual.
Where to place the rule?