arendst/Tasmota #2581: Feature request: Command for publishing retained message

definitio, 2018-04-30:
Hi,
would it be possible to add command for publishing messages with retain flag?

Jason2866, 2018-04-30:
No, there is simply not enough memory space to store this messages in the device.
There is a command for a retain flag. You can query this. Look in the wiki...
You have to take care for this in your HA System.

curzon01, 2018-05-01:
@Jason2866 - may be I'm misunderstand something but talking about MQTT a client do not need to store retain messages, this is the task for a broker

definitio, 2018-05-01:
@Jason2866, no, I want store messages in a broker. I connected sensor to gpio, set it as switch and wrote a rule that send state on changes which can be rarely so after rebooting Home Assistant state of sensor is unknown because messages aren't retained.
P.S.: Sorry for my English.

curzon01, 2018-05-02:
As I found within the code (v5.13.1a - xdrv_00_mqtt.ino line 650) the `Publish` command currently sends non-retained messages and unfortunately there is no flag for the `Publish` cmnd to send it as retained.

MYLE-01, 2018-05-02:
@definitio   I know what what you are saying been Thinking long hard about this
one idea Im thinking and Is a startup script in HA that could poll all sonoff you have
and Tasmota could send a MQTT message back of Its Status and the Setting eg
I have a my Office Light "state_topic: "stat/office/POWER" would be good if tasmota could send me the Status of the Office Light think by asking it in a MQTT message somelike   stateof/office/POWER
tasmota would send back  stat/office/POWER = ON I think HA would Then set the Switch Right.

Jason2866, 2018-05-02:
@MYLE-01 
In OpenHab the state of the devices is monitored (mqtt). The items state is updated when the result of the fired command is received. Manual updates can be done on demand or you get actual states after teleperiod time. So there is no need for retain messages

curzon01, 2018-05-04:
@Jason2866 what @MYLE-01 described was a workaround when even not using retain and even so want to have the current status of the device when OH starts. 
As I understood he solved this by using an script which polls all the status from the devices at OH startup to even get all the current status and prevent using retain

closed 2018-07-03 by stale[bot]