arendst/Tasmota #2582: It is possible to send a http command with rules ?

eeprom23, 2018-04-30:
Hey,

It is possible to send http command from a rule?

EG.
Rule ds18b20#temperatur<20 do http://sonoff/cm?cmnd=Power%20On endon

So i Can switch a Other sonoff which is far away from Sensor !

Thanks for help

ascillato, 2018-04-30:
Hi. Such command does not exist by now but it is a nice feature to add.

If you want to talk device to device, another thing you can try is using knx. 

One way could be to set a rule that turn on a local relay with that condition and by knx send that information to another Tasmota.

ascillato, 2018-04-30:
Or using a rule on a home automation system

eeprom23, 2018-04-30:
I did Not have a Home automation System and Knx 😞

ascillato, 2018-04-30:
Ok. For KNX you don't need extra hardware. It is just another protocol to communicate device to device. If you use the firmware with knx, you will have a knx configuration menu.

eeprom23, 2018-04-30:
Can i Talk with knx to another device with rules?

ascillato, 2018-04-30:
You can set on the device with the sensor a rule to turn on a relay and on knx menu you can select to send the status change of that relay to a knx address like 1.1.1.

On the other device you can set to listen to the address 1.1.1 to set the relay status.

ascillato, 2018-04-30:
There is a help on the wiki with some examples

eeprom23, 2018-04-30:
Ok, thanks for your help!

closed 2018-04-30 by eeprom23

ascillato, 2018-04-30:
When using KNX IP on Multicast, I found that some Wifi Routers drop some packages. So, I'm going to make a PR in few days to increase reliability for those wifi routers. So, test the communication between devices, like a button on one device to toggle the relay on other, and test it several times. With this simple test you will see if you could have a loss of communication.

Another thing. You should try a home automation system like [home assistant](https://www.home-assistant.io/). You just need a  Raspberry Pi and download Hassio to it. It is really cool :+1:
Also you can use [SonWeb](https://github.com/reloxx13/SonWEB) to manage Tasmota devices for upgrading the firmware.

arendst, 2018-04-30:
Or simply use a rule and mqtt publish anything you like. All you need is a mqtt broker which you should have had anyway if you're interested in Tasmota.

ascillato, 2018-04-30:
Nice.

I added this MQTT Explanation to the example 2 on rules (Execute any MQTT message when a button is pressed)

Thanks.

mcc05, 2018-10-27:
But why, if your not using a broker, this would be great. Http to control the device, so why not rule to fire http to talk to others. Simple example sonoff as thermostat, need to talk to other sonoff that controls boiler simple http command would be great.

420Dope, 2018-11-01:
I would like it too if we can send http commands. I don´t have a much complex home automation and have no time to read in and setup a MQTT, FHEM, ioBrocker or something like that.

ascillato, 2018-11-01:
@mcc05 @420dope

Http commands was already implemented. This issue is very old. 

Please, read the wiki at commands and look for websend

Thanks

420Dope, 2018-11-01:
Thanks for your answer.
I don´t mean the http commands to only control a device. I mean, for example, if the relay1 on device1 is on, the decive1 can send a http command to decive2 to turn it on or off.
Or maybe with rule E.G. how in the first post.

If this is really implemented, sorry but please give me the link because i can´t find it in the wiki...

ascillato, 2018-11-01:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

search for WEBSEND

420Dope, 2018-11-01:
Ah ok thank you. It works perfect.
Here for others who want know how it works. I set this rule:
```
rule on Power1#State do WebSend [192.168.0.20:5005] {} endon
```
Now my Shelly 1 is my doorbell and sends a http command to the [homebridge-videodoorbell](https://github.com/Samfox2/homebridge-videodoorbell) plugin and i become a push from the HomeKit App on my Phone that someone ring the doorbell.