arendst/Tasmota #2584: would like to add a HC-SR04

MYLE-01, 2018-05-01:
Im using your Tasmota on my Wemos  

can point do the track how I would add  the HC-SR04 to it

MYLE-01, 2018-05-01:
have latest version
dum question how do attach it which drop do I set

RaymondMouthaan, 2018-05-01:
Is it not clearly documented [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini#hc-sr04-ultrasonic-sensor) ?

MYLE-01, 2018-05-01:
Thanks Found it and got it work in web 
working out how to ge it to send MQTT message can it be done???

RaymondMouthaan, 2018-05-01:
Some more WIKI for you to read ;-) ... 
Checkout [Tasmota Commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) and look for TelePeriod to send ever xx seconds the sensor state to mqtt ...

Or per request send 10 to cmd/<your-wemos-topic>/STATUS this requests STATUS 10 from the device which contains the sensor info in json format and will be published to stat/<your-wemos-topic>/STATUS10 ..

MYLE-01, 2018-05-01:
SNAP just started read that
thanks again 
More Reading

closed 2018-05-01 by MYLE-01