arendst/Tasmota #2586: Hue Bridge functionality broken on my setup after upgrade to 5.13.0

acrelle, 2018-05-01:
- Hardware: [ Sonoff Basic ] and [ Amazon Echo Dot]
- [ Status 0 ]
```
07:38:13 MQT: stat/sonoff2/STATUS = {"Status":{"Module":1,"FriendlyName":["Right Bedroom"],"Topic":"sonoff2","ButtonTopic":"sonoff2","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
07:38:13 MQT: stat/sonoff2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://home.relle.uk/api/arduino/firmware.bin","RestartReason":"Software/System restart","Uptime":"0T15:34:38","StartupUTC":"2018-04-30T16:03:35","Sleep":25,"BootCount":98,"SaveCount":628,"SaveAddress":"F6000"}}
07:38:13 MQT: stat/sonoff2/STATUS2 = {"StatusFWR":{"Version":"5.13.0","BuildDateTime":"2018-04-30T17:00:56","Boot":4,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
07:38:13 MQT: stat/sonoff2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"192.168.0.200","LogPort":514,"SSId":["ACR-PRIVATE-2G","ACR-UPSTAIRS-2G"],"TelePeriod":60,"SetOption":["55CA00A9","55C18000"]}}
07:38:13 MQT: stat/sonoff2/STATUS4 = {"StatusMEM":{"ProgramSize":481,"Free":520,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
07:38:13 MQT: stat/sonoff2/STATUS5 = {"StatusNET":{"Hostname":"sonoff2-3572","IPAddress":"192.168.0.171","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:95:4D:F4","Webserver":2,"WifiConfig":3}}
07:38:13 MQT: stat/sonoff2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_954DF4","MqttUser":"relle","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:38:13 MQT: stat/sonoff2/STATUS7 = {"StatusTIM":{"UTC":"Tue May 01 07:38:13 2018","Local":"Tue May 01 07:38:13 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":0}}
07:38:13 MQT: stat/sonoff2/STATUS10 = {"StatusSNS":{"Time":"2018-05-01T07:38:13","BME280":{"Temperature":18.0,"Humidity":49.2,"Pressure":998.7},"TempUnit":"C"}}
07:38:13 MQT: stat/sonoff2/STATUS11 = {"StatusSTS":{"Time":"2018-05-01T07:38:13","Uptime":"0T15:34:38","Vcc":3.337,"POWER":"ON","Wifi":{"AP":1,"SSId":"ACR-PRIVATE-2G","RSSI":46,"APMac":"DC:EF:09:93:22:79"}}}

```

Scenario:

Device running with Emulation and  HueBridge enabled.

When trying to control the device directly using Alexa and the exposed emulated hue device, this fails on 5.13.0. Rolling back the firmware to 5.12.0 without any configuration changes, and it works again.

Console Log from 5.12.0:

07:36:01 HTP: Hue API (/wTmwJE31i4yBe11Dx5epWo0E54NjKf2m36ZkU8tb/lights/1/state)
07:36:01 HTP: Hue POST args ({"on": true})]
<device turns on>

Console Log from 5.13.0

 07:36:13 HTP: Hue API (/wTmwJE31i4yBe11Dx5epWo0E54NjKf2m36ZkU8tb/lights/1/state)
<alexa resports device not responding>

I've looked at sonoff/xplg_wemohue.ino, and the changes in 5.13.0 look nonconseqential...

arendst, 2018-05-01:
Most probably related to #2414.

As a workaround you could try to compile tasmota 5.13.0 with core version 2.3.0 which was used with 5.12.0 too.

acrelle, 2018-05-01:
@arendst I have recompiled Tasmota 5.13.0 against core version 2.3.0, and can confirm that this resolves the hue bridge issue. Good spot, thank you.

This resolves the issue for me.. as long as I remember to change the target platform in the next release..

acrelle, 2018-05-01:
Workaround was to edit platformio.ini as follows:

```
platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
```

I.e. comment out the second line, and remove comment on the first line.

IainPHay, 2018-05-01:
Same issue for me on 5.13.1 but I have not recompiled to as just upgraded ota. Switching to WeMo works though and the Alexa app lets you redefine WeMo switch as a light, so this is an alternate workaround.

acrelle, 2018-05-15:
Thank you @arendst !

closed 2018-05-15 by acrelle