arendst/Tasmota #2588: 5.13: can’t update „Upload buffer miscompare“

maciboy, 2018-05-01:
Hi,
I’ve 2 Sonoff Basic modules.
Both of them are running (unmodified) v5.12, downloaded from the release-page.
Now I wanted to upgrade both to v5.13, but upload fails with „upload buffer miscompare“.
On both 516kb are available, which seems too less for the 529kb firmware update file.

Within other issues the tip is to recompile using the right options, but as I said, I used the precompiled v5.12 without modification and now want to upgrade again using the precompiled v5.13…

Why has file size increased so much from approx 480kb of the the former version to 529kb for this version. I assume this increase is the issue :(

arendst, 2018-05-01:
See https://github.com/arendst/Sonoff-Tasmota/wiki/What's-New#otamagic

maciboy, 2018-05-01:
Now, unfortunately, I’ve a follow-up question 😢 
For one device the link http://sonoff.maddox.co.uk/tasmota/sonoff.bin worked and upgrade to 5.13 was successful.
However, for the other the „starting upgrade“ screen appeared, but now the switch seems to be completely dead… 
I assume it could not download the .bin successfully, since I think I made a typo 🦀
In the case that no correct .bin was loaded, is the switch now without any firmware?
I tried a power-cycle with no indication of live after plugging it in again.
Several button-press commands also did not show any reaction (pressed 4times, 6times, hold for >40sec)
Any chance to bring it back to live? (currently if now possibility for a serial connection)

Thanks so much!

JueBag, 2018-05-01:
I triedt to update from 5.12.0 to 5.13.0 using the openHAB rules ( via  http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin). However the update isn't performed,  the sonoffdevices are only restarted in the 5.12.0 version.

maciboy, 2018-05-01:
„If the device is unresponsive after that, try the IP only, not the refresh on actual page ( worked for me).“

As my router says, the Sonoff is not even connected to the Wifi.
Also, it does not toggle the power-state when pushing the button once…
Any ideas how I can reactivate such a 'dead' device?

JueBag, 2018-05-01:
In this case I'd try to flash it again.

maciboy, 2018-05-01:
Sine I used SonOTA until now, I have no hardware for a serial connection…
Is there an option to bring it to a state where I can flash it again using Wifi?

vogler, 2018-05-02:
I have the same Problem as @maciboy 
Previously, I updated using the image upload. This morning I tried otamagic, now my Basic and Touch are dead (no response, no LED, sitting in the dark) 😢 
No way to flash right now and the Touch is even glued to the wall 😒
Any hope?

maciboy, 2018-05-03:
So no-one is able to help us both? Or is it really that hopeless?

PierreScerri, 2018-05-23:
Hi

I am experiencing similar issues.

I have 2 Sonoff Basics with the 5.12 Tasmota.  I successfully flashed the minimal.bin followed by the allsensors.bin.  There is no sign of the sonoff on my wifi network.

Took out the switches and flashed 5.12 again via an FTDI adapter.  All went well.

Tried again with 5.14 via the FTDI and once again it would not work.

So it's back to 5.12 for now.

ascillato, 2018-06-01:
Hi

@PierreScerri , @maciboy 

Seems to a conflict with your 5.12 config that remains in flash with the 5.14 firmware.

Please, try to do a **full flash erase** and after that, flash again the 5.14

You can achieve this **full flash erase** by esptool (https://github.com/espressif/esptool)

`esptool.py --port COM5 erase_flash`

maciboy, 2018-06-04:
Today I had the time to flash the OTA-broken Sonoff Basic using my RaspberryPi as the TTL serial connector 😄 (no need to buy other HW if you have a Pi but no USB-to-TTL serial connector)

Now also this Sonoff is back in the game (but Wifi quality is really poor with this 5.14 release again!)

[image]

ascillato, 2018-06-04:
That is cool stuff! Please, can you share in the wiki how to flash with raspberry pi?

And about the wifi quality, try to compile with esplib 2.3.0 instead of 2.4.1

maciboy, 2018-06-04:
Done :)
https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi

As you can read there, I’m currently relaying on the precompiled .bin’s, so I can’t compile with other options 😞

ascillato, 2018-06-04:
Thanks a lot!

If you want to compile by yourself, all the instructions are in the wiki
also. The easiest is using arduino. The most complete is platform.io

maciboy, 2018-06-05:
@ascillato 
What I don‘t get is, why from two identical Sonoffs with same config and same firmware, one shows Signal strength <10% and barely connects, while the other has ~40% and performs fine...?

ascillato, 2018-06-05:
And about the compilation, there is also an arduino for raspberry pi. I tested and works really good.

ascillato, 2018-06-05:
If your initial issue is solved, please close this issue. Thanks a lot for your help :+1:

maciboy, 2018-06-05:
_„Seems to be a problem with the internal antenna.“_
That’s sad but I already thought this might be the case and the hardware is to blame :/

_„there is also an arduino for raspberry pi“_
What do you mean with that?

ascillato, 2018-06-05:
If you want to compile Tasmota by yourself, you can install Arduino software also in a raspberry pi by `sudo apt-get install arduino`

closed 2018-06-05 by maciboy