arendst/Tasmota #2589: Question for platformio.ini and user_config.h

roblad, 2018-05-01:
Hi,

Is it possible to read variable i.e. 
env_default = wemeos-d1-watter-PL 
or any other setting for usage
to set user_config.h
#define MODULERL 1

arendst, 2018-05-01:
This should do it but I do not support it.

closed 2018-05-01 by arendst

roblad, 2018-05-01:
Hi,

Just for confirmation, could be like that ?

platformio.ini

[platformio]

env_default = sonoff-PL
...
[common]

build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
  -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
  -DMY_LANGUAGE=pl-PL
  -DENV_NAME=${platformio.env_default}
  
and in user_config.h

....

const String ENV_V = ENV_NAME;
#define ENV ENV_NAME

#if ENV == "sonoff-PL"

...

#endif

http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags