arendst/Tasmota #2590: Sonoff T1 2CH Buttons turn ON/OFF every few seconds

rloyz, 2018-05-01:
Make sure these boxes are checked before submitting your issue - Thank you!

- [X ] What hardware you are using
Sonoff T1 2CH
- [ ] Provide the output of command ``status 0``
{"Status":{"Module":29,"FriendlyName":["ipswitch1-1","ipswitch1-2"],"Topic":"ipswitch1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Power on","Uptime":"17652T03:23:01","StartupUTC":"1970-01-01T08:00:01","Sleep":0,"BootCount":37,"SaveCount":544,"SaveAddress":"F4000"},"StatusFWR":{"Version":"5.12.0n","BuildDateTime":"2018-04-30T19:00:20","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"},"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":4,"LogHost":"fs.spider","LogPort":514,"SSId":["SPIDER","SPIDER"],"TelePeriod":300,"SetOption":["0000A001","55818000"]},"StatusMEM":{"ProgramSize":532,"Free":468,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3},"StatusNET":{"Hostname":"ipswitch1","IPAddress":"192.168.86.201","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.86.1","Mac":"60:01:94:97:36:B6","Webserver":2,"WifiConfig":3},"StatusTIM":{"UTC":"Tue May 01 11:23:02 2018","Local":"Tue May 01 12:23:02 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:29","Sunset":"20:05"},"StatusSNS":{"Time":"2018-05-01T12:23:02"},"StatusSTS":{"Time":"2018-05-01T12:23:02","Uptime":"17652T03:23:01","Vcc":3.495,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"SPIDER","RSSI":88,"APMac":"70:3A:CB:CD:4F:BF"}}}
- [ X] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Here is a log snippet after boot - Note I'm not touching the device
00:00:00 CFG: Loaded from flash at F8, Count 548
00:00:00 APP: Boot Count 38
00:00:00 Project sonoff ipswitch1-1 (Topic ipswitch1, Fallback DVES_9736B6, GroupTopic sonoffs) Version 5.12.0n-2_4_1
00:00:00 CFG: Saved to flash at F7, Count 549, Bytes 2560
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 SPIDER in mode 11N as ipswitch1...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:04 APP: Button1 immediate
00:00:04 RSL: RESULT = {"POWER1":"ON"}
00:00:04 RSL: POWER1 = ON
00:00:04 CFG: Saved to flash at F6, Count 550, Bytes 2560
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:05 APP: (UTC) Thu Jan 01 08:00:06 1970, (DST) Sun Mar 29 02:00:00 1970, (STD) Sun Oct 25 03:00:00 1970
08:00:07 WIF: Checking connection...
08:00:07 WIF: Connected
08:00:07 DNS: Initialized
08:00:07 HTP: Web server active on ipswitch1.local with IP address 192.168.86.201
08:00:07 RSL: INFO1 = {"Module":"Sonoff T1 2CH","Version":"5.12.0n","FallbackTopic":"DVES_9736B6","GroupTopic":"sonoffs"}
08:00:07 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"ipswitch1","IPAddress":"192.168.86.201"}
08:00:07 RSL: INFO3 = {"RestartReason":"Software/System restart"}
08:00:07 APP: (UTC) Tue May 01 11:25:50 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
12:25:51 HTP: Console
12:25:51 RSL: RESULT = {"POWER1":"ON"}
12:25:51 RSL: POWER1 = ON
12:25:51 RSL: RESULT = {"POWER2":"OFF"}
12:25:51 RSL: POWER2 = OFF
12:25:58 RSL: STATE = {"Time":"2018-05-01T12:25:58","Uptime":"17652T03:25:57","Vcc":3.485,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":2,"SSId":"SPIDER","RSSI":92,"APMac":"70:3A:CB:CD:4F:BF"}}
12:26:04 APP: Button1 immediate
12:26:04 RSL: RESULT = {"POWER1":"OFF"}
12:26:04 RSL: POWER1 = OFF
12:26:04 CFG: Saved to flash at F5, Count 551, Bytes 2560
12:26:10 WIF: Checking connection...
12:26:10 WIF: Connected
12:26:14 APP: Button1 immediate
12:26:14 RSL: RESULT = {"POWER1":"ON"}
12:26:14 RSL: POWER1 = ON
12:26:14 CFG: Saved to flash at F4, Count 552, Bytes 2560
12:26:30 WIF: Checking connection...
12:26:30 WIF: Connected

***************************************
The buttons just keep turning on an off. I can toggle them by touching. But they randomly keep happening.

Note I this is only the touch board and 8255/Wifi, there is no Switch board connected and it is plugged into a 5V USB powered connector via the interboard connector - which typically supplies 5V when the unit is connected to the mains.

I don't want the switch circuitry. My plan was for a low power physical switch that would just toggle and send a message to my MQTT server, with the plan to use this switch to turn on/off another power circuit that is a long way away (via interpreting the MQTT message and turning on a remote Sonoff Basic). Eventually this low power switch was to be "ruggedised" for outdoor use and the low power meant that heat dissipation wouldn't be a problem once hermetically sealed. The plan was for the switch to be located outdoors near the house, but control a bunch of outdoor garden light circuits.

Unfortunately I'm failing at the first hurdle. It just continually toggles when left alone, every 10-15 seconds or so.

Frogmore42, 2018-05-01:
Touch switches can be "touchy" to get working correctly.  I am sure the T1 was not designed to be used outdoors and would likely have problems in that type of environment.  Are you planning on powering this by batteries?  If so, this is not a good choice, because unless you go to a lot of trouble it needs at least a half watt of power.  If you will have power available at the location, I would use a wemos D1 and a physical switch instead.  It will likely be more reliable.

rloyz, 2018-05-01:
It's currently failing indoors. The plan was to power it at 5V into the 2 pins in J2 that supply 5V from the transformer/switch board via a USB power plug. It's currently set up that way on a desk.

rloyz, 2018-05-02:
Solved. It's just bad power - damn those dirty USB wall warts!
Seems that the board really needs clean power and the transformer/switch side provided a very clean 5V to the system. Hooking up to 3 AA batteries in series provided very clean power and bingo, no more crazy random switches. I'll add some a filter on the 5V input and see if that fixes the USB supply enough to have it work.

closed 2018-05-02 by rloyz