arendst/Tasmota #2592: Bulgarian version not working

brandyhs, 2018-05-01:
Hello, i update the version to 5.13 classic and then i see there is a bg version.I updated it via OTA and my sonoff basic stop working. i tried to flash this firmware via esptool and not working again.Please fix it

w2cker, 2018-05-01:
I also have problems with the dutch verion. Flashing the minimal version is working. The dutch precompiled bin version is not working. I tried a old sonoff and a new 2.0 version. Flashing back the old firmware restores the sonoff.

arendst, 2018-05-01:
See update above

arendst, 2018-05-01:
Dutch no problem either.

[image]

w2cker, 2018-05-01:
I have tried flashing again. Now with the update 5.13.1 version. OTA to minimal is fine. Second step to 5.13.1 is flash fine but after reboot Wifi is not working. The led is flashing about twice every 2 seconds, and a Sonoff wifi_AP is sometimes visible, however unable to connect. Flashing by serial cable also has no effect.

brandyhs, 2018-05-02:
my problem is the same as w2cker's

closed 2018-07-01 by stale[bot]