arendst/Tasmota #2593: Sonoff 4CH Pro Module - MQTT parameters - Topic value change hostname

fpalinf, 2018-05-01:
When I change the value of the HTML entry Sonoff 4CH Pro Module - MQTT parameters - Topic
it changes the hostname.
Is this the expected behaviour ?

(also, when I change the hostname in Wifi Parameters - Hostname from %s-%04d to static value and press save. It does not save the value)

git be36218b846099fee336d03df30c5f3684fc3c42

arendst, 2018-05-01:
Do not observe

fpalinf, 2018-05-01:
The only change I made from the git code was the update in
`sonoff/user_config.h
  #define CFG_HOLDER 0x20180426
  #define STA_SSID1 "ssid"
  #define STA_PASS1 "ssidpwd"`
and `platformio.ini` uncommented: `env_default = sonoff`
according to the page [Upload wiki guide](https://github.com/arendst/Sonoff-Tasmota/wiki/Upload)

From what steps could come the difference between what I observe and the expected behaviour ?

closed 2018-06-30 by stale[bot]