arendst/Tasmota #2595: Question regarding OTAMagic!

JueBag, 2018-05-01:
On the WhatsNew page under OtaMagic is stated:

> This process is now automated if used with an external OTA server.
> 
> Tasmota now tries to load the requested image and if it notices that the image won't fit it will load the minimal version first which in turn will load the requested final image.  This is OtaMagic.

My English might be a bit OFF, as far as I understood when trying to update from an external OTA server ( like http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin) the process would be running automatically over both stages. I.E. giving the command "upgrade/1" after setting the correct otaURL would start the automated process.
Since this was stated under the 5.12.0 section I would have assumed this to be working for the update from 5.12.0 to 5.13.0. However it did not. 

Where is my error?

JueBag, 2018-05-01:
Sorry, that was a misspelling. The rule using the command didwork for upgrading from 5.11 to 5.12.
So, your answer is that the automatic upgrade from 5.12 to 5.13 should have worked?

arendst, 2018-05-01:
It depends where you get your firmware from. I do not own Maddox so I do not know if the latest images are available there.

Using my local OTA server it works as expected.

JueBag, 2018-05-01:
Understood!
And a big THANK YOU for this software!

closed 2018-05-01 by JueBag