arendst/Tasmota #2598: SwitchMode and SwitchTopic more Confusion

mcc05, 2018-05-01:
So I have a sonoff basic, toggle switch on GPIO14, set as 09 Switch1. 
If I use the default switchmode=0 and switchtopic=0, all is well, but the only way to get the state of the GPIO14 switch is the SENSOR value which is the wrong way round as when the physical switch is ON the GPIO is ground and the sensor states OFF and vice versa.

So if I set the switchmode=2 this corrects this problem but the relay does not toggle if follows the GPIO14 switch value.  So I tried setting the switchtopic etc. all don't seem to give me the result i would like.

So is it possble to report physical switch state (invert GPIO14 state), and if this toggles simply toggle the relay.

Mark

Jason2866, 2018-05-01:
Could you explain, why it is interesting for you to know the status of the extern switch?
You have a reliable info from the tasmota device if it is on or off.

Frogmore42, 2018-05-01:
I think you want switch mode 1 and switchtopic 1
You have a toggle switch so this will cause it to match what the switch is showing.  Even if you want the switch to always toggle, you should use switchtopic 1 to get the mqtt messages when the switch is switched.

closed 2018-06-30 by stale[bot]