arendst/Tasmota #2599: Wrong name for 5.13.1

0n3-70uch, 2018-05-01:
Hello, just a beauty mistake. The version 5.13.1 shows 5.13.0a.

Need to change this:
`#define VERSION                0x050D0001   // 5.13.0a`

to this:
`#define VERSION                0x050D0100   // 5.13.1`

Greetings Fabi

0n3-70uch, 2018-05-01:
Oh no. Still in the official releases:
https://github.com/arendst/Sonoff-Tasmota/archive/v5.13.1.zip

0n3-70uch, 2018-05-01:
Ok there is another problem. The platformio.ini from the master branch is different to these:
https://github.com/arendst/Sonoff-Tasmota/releases

arendst, 2018-05-01:
Yes. I know what I did wrong. Just removed 5.13.1 (5.13.0a) and re-released 5.13.1 as the real one.

Sorry.

0n3-70uch, 2018-05-01:
Ah ok. Thank you.
The new version is running very nice on five devices at the moment.

closed 2018-05-01 by 0n3-70uch