arendst/Tasmota #2600: Sonoff TH16 pin locations?

demian85, 2018-05-01:
Hi, sorry if this sounds like a stupid question, but I cannot seem to find the schematic for the Sonoff TH16 and I don't know for sure where are the Tx/Rx pins.

I assume this is the one I'm looking for: https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations

But after searching Google, I found another pic that says otherwise.

I keep getting this error when trying to upload firmware:
```
Auto-detected: /dev/cu.SLAB_USBtoUART
Uploading .pioenvs/sonoff/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```

After reading a lot of issues, the only thing that could be wrong is that Tx/Rx pins are switched.

Can someone please confirm and clarify this in the wiki?

Thanks.

demian85, 2018-05-01:
> Hope this helps

Thanks. Seems that the wiki is okay, I just tried again and it worked -_-

closed 2018-06-04 by ascillato

Fvrajo, 2019-03-08:
https://www.itead.cc/wiki/images/3/39/Sonoff_TH10A%2816A%29_schmatic.pdf

Fvrajo, 2019-03-08:
Pin | Sonoff | USB
-- | -- | --
1 | VCC | 3V3
2 | RX | TX
3 | TX | RX
4 | GND | GND

jiangqian, 2019-11-21:
The schematic looks with some issue:https://www.itead.cc/wiki/images/3/39/Sonoff_TH10A%2816A%29_schmatic.pdf

The 2.5 mm jack ping assignment from tip to the pole next to the handle is: scl, sda, GND, 3.3 V but the schematics is item2, 3.3v, item1, GND.

Does anyone know if there is updated schematic?