arendst/Tasmota #2602: firmware upgrade OTA problem

chris-008, 2018-05-01:
Hi,
Referring to a previous issue #1710, i still have a similar problem:
while trying to upgrade a Sonoff TH10 module , flashed with  a minimal 5.10.0 version, I get next message in the web interface:
Upgrade started ...
Device will restart in a few seconds

This screen stays like this for minutes and I hear the module clicking every now and then but nothing much happens. When I press the main menu button, the module doesnt get recognized anymore via the web browser and I have to reflash all over.

any ideas what could go wrong?
I am using http://sonoff.maddox.co.uk/tasmota/sonoff.bin  as an OTA link

since http://domus1:80/api/arduino/sonoff.ino.bin is not working.

 thanks for your feedback!

chris-008, 2018-05-02:
Thanks for the update. 
I just copied all parameters from my last working veriosn 5.10.
However, with this new 5.13.1 firmawre I get a COM port problem:
Use manually specified: COM5
error: cannot access COM5

When i do the same with ver 5.10, all is working fine (port gets selected automatically)
Where should I change this setting in ver 5.13.1?

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_speed = 115200
upload_speed = 512000
upload_resetmethod = nodemcu
upload_port = COM5  //tried all from 1 to 5

Frogmore42, 2018-05-03:
Don't specify the com port or figure out which com port it really is.  On Windows you can use Device Manager to list all the com ports on your PC.

chris-008, 2018-05-03:
Thanks for your feedback. I use a MacBook. Wit a precious firmware Version (5.10.0) , the same port (5) is auto detected and works just fine?

Frogmore42, 2018-05-04:
I don't have a Mac so I don't know what error it gives if the com port is already open.  On Windows, the tool automatically detects the right com port on all my PCs.  I can also use Device Manager to see all the com ports.  I don't know if the Mac has a similar tool.

I don't believe the FW makes a difference on finding or opening the com port.  So either the com port is in use (did you have the serial monitor on earlier) or it is something else.

chris-008, 2018-05-05:
This is what I found:

FT232R USB UART:

  Productcode:	0x6001
  Fabrikantcode:	0x0403  (Future Technology Devices International Limited)
  Versie:	6.00
  Serienummer:	A50285BI
  Snelheid:	Maximaal 12 Mb/sec
  Fabrikant:	FTDI
  Locatiecode:	0x14200000 / 12
  Beschikbare stroom (mA):	500
  Vereiste stroom (mA):	90
  Extra stroom in werking (mA):	0

So no specific port mentioned.
Isn't there a way to have the port detected automatically, as in previous builts?

chris-008, 2018-05-05:
in eg. 5.10 (working version) there is this code:

:[env:sonoff]
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu

In the 5.13 version:
; *** Upload Serial reset method for Wemos and NodeMCU
;upload_speed = 115200
upload_speed = 512000
upload_resetmethod = nodemcu
upload_port = COM5

uncommenting the last two lines compared to the 5.10 built creates new errors.

Frogmore42, 2018-05-05:
Since you don't have a Wemos/NodeMCU, you don't need to uncomment that line.  I never specify a COM port and it always finds it, so I don't believe you need to uncomment the COM port line either.  512000 is also way too fast for most adapters that use long wires.  So, make it look like it did before, ie comment everything except the line that has the upload speed of 115200.

chris-008, 2018-05-08:
So the 5.13.1 upload worked like a charm.
The update procedure however is still a bit of a mystery...

After I tried to upgrade the firmware from minimal to standard sonoff firmware through the web interface (after resetting the module first), I get the update screen hanging indefinitely 

[image]

and got next readout from Atom, indicating a WIFi connect problem:

[image]

The module is an an infinite loop searching for wifi, with LEDS flashing in intervals.

I have WIFI_MANAGER as wifi config tool set.

chris-008, 2018-05-10:
Wifi manager doesn't seem to work either, 192.168.4.1. is not showing the UI.

Frogmore42, 2018-05-10:
Some people have a problem with core 2.4.0 (which the FW you built is using).  You might try rebuilding the FW with 2.3.0.  There should be a line near the top that tells PlatformIO which core to use.  I believe there are several lines there (to show the possibilities).  Uncomment the one with 2.3.0 and comment the one that is currently uncommented.

chris-008, 2018-05-10:
Do you mean this part:
[image]

Frogmore42, 2018-05-11:
Yes, that is the part, comment the one that doesn't specify and uncomment the one with 2.3.0

chris-008, 2018-05-24:
Hi Frogmore,  thanks again for the feedback, will try soon.

chris-008, 2018-05-28:
Hi, So I changed as per your proposal:

[common] ; ************************************************************
; *** Esp8266 core for Arduino version
platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266@1.7.0  ; v2.4.1
;platform = espressif8266

Still the module hangs in upgrade mode. 
Every 45s. I hear the module clicking but no upgrade still.

Frogmore42, 2018-05-28:
If you have a serial connection, there is no need to upgrade.  Just flash the full version, not the minimal one.  The minimal one is for use with Over The Air FW update.

Include the log if you have any more problems.

chris-008, 2018-06-02:
No, it doesn't help

closed 2018-08-01 by stale[bot]

mikeyford, 2018-12-03:
Quick note for any mac OSX users that come across this, as I did:

Replace `COM5` in platformio.ini with the full path to your serial-usb device. You can find the path by entering `ls -l /dev/tty.*` in the terminal. For example, I changed my entry to this:
`upload_port = /dev/tty.usbserial-A602286D` and was good to go.