arendst/Tasmota #2604: Kompilation Issue 5.13.1

helmar74, 2018-05-02:
Make sure these boxes are checked before submitting your issue - Thank you!

If I want to compile Tasmota 5.13.1 I get the following compile error:

In file included from C:\temp\SONOFF_Tasmota\Sonoff-Tasmota-v5.13.1\sonoff\sonoff.ino:37:0:

sketch\sonoff_post.h:41:24: fatal error: esp-knx-ip.h: No such file or directory

 #include <esp-knx-ip.h>

                        ^

compilation terminated.

exit status 1
Fehler beim Kompilieren für da

Does anyonde know, what to do?

Thanks a lot

- [SONOFF BASIC ] What hardware you are using

helmar74, 2018-05-02:
Closed, missing libaries in Sketchbook ;-)

closed 2018-05-02 by helmar74

NischalBhandari, 2018-05-10:
dear sir how did you solve the problem , my sketchbook has all the needed libraries . What specific library is there to add ?

NischalBhandari, 2018-05-11:
worked for me by copying all the libraries from tasmota file to libraries in c:\program Files\arduino\libraries

cpyandy, 2018-06-09:
@helmar74 
I have got a similar issue, i thought the issue is caused by missing library, so I did what's mentioned by @NischalBhandari , but it's not working for me, am I copying the wrong lib file to wrong location??

mrohner, 2019-11-02:
This is not a good solution as I do have same libraries but older releases in the libraries directory already. I do not want to upgrade them, who knows if the new versions work with my older programs

nigelhewitt, 2020-08-20:
I'm having the same problem.
I tried copying Tosmota-master\lib to Arduino\lib so now the 'missing' file exists in two places but it still didn't find it.
Just a pointer to where to look for an error would help. I know VS2019 and eclipse a bit but I'm new to this IDE