arendst/Tasmota #2606: Any way to "discover" sonoff tasmota devices on WiFi only (no Internet)

anoop-r-kulkarni, 2018-05-02:
I have been successfully using the MQTT discovery and scanning status messages helps me build a list of devices currently powered on and connected to the WiFi. I am looking for solutions to see if this equivalent discovery is possble in "no Internet" scenario (that is WiFi only). If I am using an Android app for this, is my only solution to build a local MQTT broker in the app or is there a simpler way of discovering connected devices that are powered on at a given instant?

Thanks and regards
~anoop

Jason2866, 2018-05-02:
Take a look in your DHCP Server. Normally this is your router. You can see here which device has a lease and the IP and in most cases the name of the device...

enesbcs, 2018-05-04:
There are a network scanner named ESPFinder-python which shows active Tasmota and ESPEasy devices on the LAN. Mainly for linux, partially works for windows. For android i have no idea.
[image]

anoop-r-kulkarni, 2018-06-02:
Thanks.. did it the brute force way by pinging each IP for 255 devices in the subnet..

closed 2018-06-02 by anoop-r-kulkarni

ascillato, 2018-06-02:
Hi,

You can use also https://github.com/reloxx13/tasmoAdmin

anoop-r-kulkarni, 2018-06-02:
thank you.

I am stll trying to figure out an optimal setup. We have 30+ TH16 
connected to ducted Air Conditioners at a faciltiy and other than MQTT, 
trying to figure out what is the most optimal way of accessing all the 
devices with the local WiFi

Will explore TasmoAdmin as well.

Thanks a lot..

Best,
~anoop

------------------------------------------------------------------------

*From:* Adrian Scillato
*Sent:* Saturday, June 02, 2018 10:00AM
*To:* Sonoff-Tasmota@noreply.github.com
*Cc:* anoop.r.kulkarni@gmail.com, state_change@noreply.github.com
*Subject:* Re: [arendst/Sonoff-Tasmota] Any way to "discover" sonoff 
tasmota devices on WiFi only (no Internet) (#2606)

Hi,

You can use also https://github.com/reloxx13/tasmoAdmin