arendst/Tasmota #2608: Unable to compile with Rules not selected

jeylites, 2018-05-02:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

ESP: V2.4.1
Hardware: Sonoff Touch T1 UK
Arduino: 1.8.5 OSX

```
// -- Rules ---------------------------------------
//#define USE_RULES                                // Add support for rules (+4k4 code)

/Users/R/Documents/Arduino/Sonoff-Tasmota5.13.1a/sonoff/support.ino: In function 'void AdcEvery50ms()':
support:1412: error: 'RulesProcess' was not declared in this scope
       RulesProcess();
                    ^
exit status 1
'RulesProcess' was not declared in this scope
```

Leseratte10, 2018-05-02:
Looks like support.ino#L1412 `RulesProcess()` needs to be wrapped in `#ifdef USE_RULES` - `#endif`-blocks.

jeylites, 2018-05-03:
@leseratte10

What does that mean? :)

Leseratte10, 2018-05-03:
That means there is (was) an error in the Tasmota code and I just described what needs to be changed to fix this. The newest version (e6e1263) fixes the bug, so just redownload and try again.

jeylites, 2018-05-03:
thank you for the explanation! Just saw the fix. Thanks Arend!

closed 2018-05-03 by arendst