arendst/Tasmota #2611: 5.13.0 or 5.13.1: not flash or upgrade

RWGaMeR, 2018-05-02:
Good morning,
I tried the firmware (5.13.0 - 5.13.1) on two Sonoff Basic, but only give problems because they do not remain active or do not remain active and the MQTT protocol from many problems.

I wait for these problems to be solved

reloxx13, 2018-05-02:
change and try 2.3.0 framework

RWGaMeR, 2018-05-02:
Do i have to do?

RWGaMeR, 2018-05-02:
@reloxx13 Do i have to do?

reloxx13, 2018-05-02:
build the firmware yourself (suggest to use atom and platformio ([Guide](https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build)).

change to 
`platform = espressif8266@1.5.0  ; v2.3.0`
in the platformio.ini file

i noticed wifi disconnects with 2.4.0 and 2.4.1, check your router log to get the disconnects. mine was flooded with disconnects every few minutes.

RWGaMeR, 2018-05-02:
@reloxx13  Thanks, I'll try tomorrow

nunoronconcouto, 2018-05-02:
i downgrade to 5.13.0...

RWGaMeR, 2018-05-03:
@nunoronconcouto 
I have already downgraded to 5.13.0, but it does not work well the same, I had to go back to 5.12.0

pReya, 2018-05-03:
Same problem here, many WiFi disconnects with 5.13.1 (precompiled binary version).

alton76, 2018-05-03:
> @pReya <https://github.com/pReya>
> From what I understand is really the framework, today the text change
> framework and then I tell you if it works!

Give Tasmota 5.13.1a and ESP V2.4.1 a try. I've been running for two days
straight, no complains.

Previously I was running Tasmota 5.12.1i and V2.4.0 and V2.30, lots of
problems, frequent disconnection, unable to reach web server.

RWGaMeR, 2018-05-03:
@alton76 
I with the 5.12.0 firmware I had no problems, only after I put 5.13.1 I had a lot of problems and I came back to 5.12.0, today text as said that they suggested

alton76, 2018-05-03:
What ESP Version?

RWGaMeR, 2018-05-03:
@alton76 

I use a basic sonoff that has esp8266

RWGaMeR, 2018-05-03:
@reloxx13 
I tried as you suggested but I have not solved the problem

closed 2018-07-02 by stale[bot]