arendst/Tasmota #2613: 5.13.1 Incorrect pulsetime when it was triggered by rule

wongnam, 2018-05-03:
Make sure these boxes are checked before submitting your issue - Thank you!
- [x] What hardware you are using
- [x] Provide the output of command ``status 0``
07:58:36 CMD: status 0
07:58:36 MQT: stat/livingroom/STATUS = {"Status":{"Module":39,"FriendlyName":["Living Room Fan","Living Room Light 1","Buzzer","For Test"],"Topic":"livingroom","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
07:58:36 MQT: stat/livingroom/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:41:18","StartupUTC":"2018-05-03T00:17:18","Sleep":0,"BootCount":18,"SaveCount":830,"SaveAddress":"FA000"}}
07:58:36 MQT: stat/livingroom/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-02T21:05:24","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:58:36 MQT: stat/livingroom/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Multimedia Wi-Fi Network","Wongnam"],"TelePeriod":300,"SetOption":["00808009","55C18000"]}}
07:58:36 MQT: stat/livingroom/STATUS4 = {"StatusMEM":{"ProgramSize":523,"Free":480,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
07:58:36 MQT: stat/livingroom/STATUS5 = {"StatusNET":{"Hostname":"livingroom-2739","IPAddress":"192.168.12.133","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.12.1","Mac":"60:01:94:98:0A:B3","Webserver":2,"WifiConfig":3}}
07:58:36 MQT: stat/livingroom/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.126","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_980AB3","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:58:36 MQT: stat/livingroom/STATUS7 = {"StatusTIM":{"UTC":"Thu May 03 00:58:36 2018","Local":"Thu May 03 07:58:36 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":7,"Sunrise":"05:33","Sunset":"18:05"}}
07:58:36 MQT: stat/livingroom/STATUS10 = {"StatusSNS":{"Time":"2018-05-03T07:58:36","Switch1":"ON","Switch2":"ON"}}
07:58:36 MQT: stat/livingroom/STATUS11 = {"StatusSTS":{"Time":"2018-05-03T07:58:36","Uptime":"0T00:41:18","Vcc":3.160,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":80,"APMac":"6C:70:9F:E4:88:20"}}}

I am using Sonoff Dual R2 with Tasmota firmware version 5.13.1
the issue i am facing is PulseTime for Power3 incorrect pulsetime when it was triggered by "rule on Power1#State=OFF do Power3 ON endon", it only turn the relay on a very short timing.

if i directly press the button on webpage it will have the correct Pulsetime.

Thanks.

curzon01, 2018-05-03:
`PulseTime3 5` means 0,5 sec on because values below 112 are in 0.1 sec steps.
Is that what you mean with "very short time" than this is correct.

- Do you use sleep? What's your `Sleep` value (enter command `Sleep` and give us the result)?
- instead of clicking on "Toggle 3" use Web Interface "Console" and type the power command `Power3 ON`

wongnam, 2018-05-03:
Yes, i am aware that 5 is 0.5 second. 
my action taken as below:
- use console and type command ON for Power3: result pulsetime is correct for 0.5s
- use MQTT send comnand "ON" to Power3: it also has correct result for 0.5s
- use "Rule on Power1#State=OFF do Power3 ON endon" : the actual plusetime is just about 0.1s

arendst, 2018-05-03:
Pls provide an overview of your configured GPIOs as a dual normally has just two relays, not four.

I expect you are using GPIO0 to control a relay which might fail as your last status is a Hardware Watchdog...

wongnam, 2018-05-04:
This is my Sonoff Dual R2 configurations
For easy, you can treat as a 2 relays with 2 more relays at GPIO3 and GPIO1.
[image]

- Pulsetime3 5 or even longer value but it still performs incorrectly.
- SerialLog 0
- Emulation 2
[image]

Thanks

wongnam, 2018-05-04:
Hi Arendst,

For easy to imaging the issue, i have configure my Sonoff Dual R2 to a very simple.
you can easy to see that my pulsetime set to 40 (=4sec) and the rule perform the Power2 On about 4sec on console screen but the physical relay2 on Sonoff click about 0.2sec or 0.5sec. =>The problem is that I want to say here.

**Below is my console info FYR:**
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_980AB3, GroupTopic sonoffs) Version 5.13.1-2_3_0
00:00:00 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as sonoff-2739...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff-2739.local with IP address 192.168.12.133
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff/POWER = 
00:00:05 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Dual R2","Version":"5.13.1","FallbackTopic":"DVES_980AB3","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-2739","IPAddress":"192.168.12.133"}
00:00:05 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
00:00:05 MQT: stat/sonoff/POWER1 = OFF
00:00:06 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/sonoff/POWER1 = OFF
00:00:06 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
00:00:06 MQT: stat/sonoff/POWER2 = OFF
09:30:19 MQT: tele/sonoff/STATE = {"Time":"2018-05-04T09:30:19","Uptime":"0T00:00:13","Vcc":3.160,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":90,"APMac":"6C:70:9F:E4:88:20"}}
09:30:19 MQT: tele/sonoff/SENSOR = {"Time":"2018-05-04T09:30:19","Switch1":"ON","Switch2":"ON"}
09:30:23 CMD: switchmode1 3
09:30:23 MQT: stat/sonoff/RESULT = {"SwitchMode1":3}
09:30:37 CMD: **pulsetime2 40**
09:30:37 MQT: stat/sonoff/RESULT = {"PulseTime2":"40 (Active 0)"}
09:30:52 CMD: rule on power1#state=off do power2 on endon
09:30:52 MQT: stat/sonoff/RESULT = {"Rule":"OFF","Once":"OFF","Rules":"on power1#state=off do power2 on endon"}
09:30:58 CMD: rule on
09:30:58 MQT: stat/sonoff/RESULT = {"Rule":"ON","Once":"OFF","Rules":"on power1#state=off do power2 on endon"}
09:31:07 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:31:07 MQT: stat/sonoff/POWER1 = ON
**09:31:10** RUL: POWER1#STATE=OFF performs "power2 on"
**09:31:10** MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
**09:31:10** MQT: stat/sonoff/POWER2 = ON
**09:31:10** MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
**09:31:10** MQT: stat/sonoff/POWER1 = OFF
**09:31:14** MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
**09:31:14** MQT: stat/sonoff/POWER2 = OFF

09:31:41 CMD: status 0
09:31:41 MQT: stat/sonoff/STATUS = {"Status":{"Module":39,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:31:41 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:35","StartupUTC":"2018-05-04T02:30:06","Sleep":0,"BootCount":3,"SaveCount":14,"SaveAddress":"F6000"}}
09:31:41 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-02T21:05:24","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:31:41 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Multimedia Wi-Fi Network","Wongnam"],"TelePeriod":300,"SetOption":["00808009","55818000"]}}
09:31:41 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":523,"Free":480,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
09:31:41 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2739","IPAddress":"192.168.12.133","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.12.1","Mac":"60:01:94:98:0A:B3","Webserver":2,"WifiConfig":3}}
09:31:41 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.126","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_980AB3","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:31:41 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri May 04 02:31:41 2018","Local":"Fri May 04 09:31:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":7,"Sunrise":"05:33","Sunset":"18:05"}}
09:31:41 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-05-04T09:31:41","Switch1":"ON","Switch2":"ON"}}
09:31:41 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-05-04T09:31:41","Uptime":"0T00:01:35","Vcc":3.160,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":88,"APMac":"6C:70:9F:E4:88:20"}}}

[image]

[image]

arendst, 2018-05-04:
Thnx for explanation!

I see rule ``on power1#state=off do power2 on endon``

rule state does not recognize on or off but only numbers (0 = off, 1 = on) so please try again using a rule like ``on power1#state=0 do power2 on endon``

arendst, 2018-05-04:
Ah, I see. It fails on me too.

A workaround that works for me is a rule like ``on power1#state=0 do backlog delay 1;power2 on endon``. This inserts a 0.1 second delay before power2 is turned on.

wongnam, 2018-05-04:
Hi Arends,

Your advice is working for my case. Thanks
0.1sec delay is no probem for me, but i looked in the console log i saw delay value is number “2”.

21:09:13 RSL: POWER1 = ON
21:09:15 RUL: POWER1#STATE=0 performs "backlog delay 1;power2 on"
21:09:15 RSL: RESULT = {"Backlog":"Appended"}
21:09:15 RSL: RESULT = {"POWER1":"OFF"}
21:09:15 RSL: POWER1 = OFF
21:09:15 RSL: RESULT = {"Delay":2}
21:09:15 RSL: RESULT = {"POWER2":"ON"}
21:09:15 RSL: POWER2 = ON
21:09:16 RSL: RESULT = {"POWER2":"OFF"}
21:09:16 RSL: POWER2 = OFF

arendst, 2018-05-04:
Ah, right that's because I didn't want to make delay to small for backlog commands....

I suggest in your case just leave the delay out as I suggested in the (updated) previous answer.

wongnam, 2018-05-04:
Hi

Fortunately, I also can still use Power1#State=ON intead of number 0, for me it is a benefit due to it is easy to read and remember. Thanks 

21:26:21 RSL: RESULT = {"Rule":"ON","Once":"OFF","Rules":"on power1#state=OFF do backlog power2 ON endon"}
21:26:26 RSL: RESULT = {"POWER1":"ON"}
21:26:26 RSL: POWER1 = ON
21:26:28 RUL: POWER1#STATE=OFF performs "backlog power2 ON"
21:26:28 RSL: RESULT = {"Backlog":"Appended"}
21:26:28 RSL: RESULT = {"POWER1":"OFF"}
21:26:28 RSL: POWER1 = OFF
21:26:28 RSL: RESULT = {"POWER2":"ON"}
21:26:28 RSL: POWER2 = ON
21:26:28 RSL: RESULT = {"POWER2":"OFF"}
21:26:28 RSL: POWER2 = OFF

arendst, 2018-05-04:
But be careful as Power1#State=ON will do the same as Power1#State=OFF as in both cases it is not a number and therefore is substituted by 0.

wongnam, 2018-05-04:
Yay! I get your point. I go with number now. I understand the difference.

Thanks

arendst, 2018-05-06:
With the just released fixes the need for backlog is gone. It also adds tests for state=On and/or state=Off