arendst/Tasmota #2616: extending Tasmota

timwilky, 2018-05-03:
I have added LCD support to my Tasmota environment. But in doing so I had to add a new command to manage the I2C address.

The issue I experienced was that there is a conditional compilation whilst defining the kTasmotaCommands array.
#ifdef DEBUG_THEO
  "|" D_CMND_EXCEPTION
#endif

Therefore if people make the logical mistake I made and add their new commands at the end of the TasmotaCammands enumeration and the kTasmotaCommands array, they become out of sync if the DEBUG_THEO flag is not in place!

arendst, 2018-05-03:
See branch development_display for a better solution.

arendst, 2018-05-03:
See #2557 too

ascillato, 2018-06-02:
Hi,

If your issue was solved, please, close it. Thanks :+1:

closed 2018-06-04 by ascillato