arendst/Tasmota #2617: italian translation

fragolinux, 2018-05-03:
hi
some suggestions for the italian translation file ( https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/language/it-IT.h )

as the file is stated to be in UTF8 format, all the letters that are present in the form "a' " (5 occurrences)  and "e' " (1 occ.) should be changed to the correct ones: "à" and "è"... and the only "u' " should be just "u" (for "menu" as in every other language...), even if both menù and menu are correct, better to go for the international version "menu"...

and of course thanks to Gennaro Tortone for the translation :)

reloxx13, 2018-05-03:
Heya,

you can edit the file yourself in your browser by clicking the pencil icon in the top right corner.
After that you can make a pull request with your changes 👍

fragolinux, 2018-05-03:
didn't know that :)
done, thanks! :)

closed 2018-05-03 by fragolinux