arendst/Tasmota #2620: Error in timer when set using mosquitto_pub

biotecov, 2018-05-03:
Hello, I am Ricardo from Spain, new to Tasmota, and already in love with it. 

I flashed a Sonoff S20 with Tasmota 5.13.1 and installed mosquitto 1.4.15 on a local linux machine. Wonderfull! However, I found what seems to be a bug while trying to set timers using mosquitto_pub.

The following:
mosquitto_pub -h localhost -t cmnd/mytopic/timer1 -t '{"Arm":1,"Mode":2,"Time":"-00:20","Window":10,"Days":"DLMXJVS","Repeat":1,"Output":1,"Action":1}'
should set power to be turned on every day 20 minutes BEFORE sunset (plus/minus 10 random mins)-

That gives a wrong result in two components as you can see either using mosquitto_sub or checking the timer on the web interface.

Using mosquitto_sub, you actually obtai:

{"Timer1":{"Arm":1,"Mode":2,"Time":"00:20","Window":10,"Days":"1111111","Repeat":1,"Output":1,"Action":0}}
 
That is, Time and Action return wrong values:

Action is actually set to 0 (off), instead of 1 (on). In fact, if you publish "Action":2, you actually obtain "Action":1, and son on (weird, one unit less!).

And the "Time" parameter cannot accept negative values. But if I publish a value greater than 11, then I actually set a negative time.

For example, "Time":"13:20" gives "Time":"-01:20", and the minus sign is also visible on the web interface. However, if I publish "Time":"12:20", mosquitto_sub shows "Time":"00:20" (without the expected minus sign). The web interface shows that you had actually obtained -00:20 in this case.

In short, if seems that the JSON string is either missinterpreted or missrepresented on the timer command.

Best regards, and congrats for such a nice software.

Ricardo.

arendst, 2018-05-04:
Nice write-up. I will investigate.

biotecov, 2018-05-04:
Woaw! How a fast response time!

Tested and solved, thanks, but testing took me a lot of time for a different reason: I was unable to update the firmware. After reading other posts, I found the reason, I was compiling with the default platformio.ini, and until I changed to use "platform = espressif8266@1.5.0  ; v2.3.0", the resulting firmwares were totally unstable. Apparently, many people had the same problem with Tasmota 5.13.x. Maybe  you could change the default selected platform in the provided platformio.ini until this issue is solved.

 Thanks, and regards,
Ricardo.

closed 2018-06-02 by ascillato