arendst/Tasmota #2623: Cannot Reprogram Sonoff T1 R2 EU Touch

oblansundar, 2018-05-04:
Hello,

I am using the firmware mentioned here - https://github.com/arendst/Sonoff-Tasmota to flash the Sonoff T1 R2 EU Touch. So far I have used this firmware to flash Sonoff Basic and Sonoff POW and flashing worked fine without any issues. But for Sonoff T1, I am unable to flash it. The error message that i get is:

**[upload] Error -1
(or)
espcomm_sync failed**

**I tried these steps for flashing:**

- https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU, I entered boot mode (1,7) 
verified in the serial console - putty with COM5 and baud rate as mentioned. I attached the serial FTDI adapter and flashed it, but  i keep getting the above error.
 
- I also tried by pullilng R19 (pin 2) to ground as mentioned in https://github.com/arendst/Sonoff-Tasmota/issues/1399 but even then I get the same error. 
 

So, could you please let me know how to bypass this error message and flash the device? I have attached the image of Sonoff T1 R2 EU by this post.

[image]
[image]

Jason2866, 2018-05-04:
Is the USB serial converter chip a FTDI? Maybe it works not with the T1.
Reason: The FTDI board may not deliver enough current. Try one with a CH340 chip.

znanev, 2018-05-04:
@Jason2866 your converter is fine - it is CP2102 based and it should be able to provide enough current to the ESP chip. No need to change it for a CH340 converter.

Can you try limiting the baud rate in platformio.ini file:

upload_speed = 115200

Also, are you sure that your USB converter is indeed connected to COM5? You can check this in Device Manager and change the value in platformio.ini if needed.

oblansundar, 2018-05-04:
@znanev Thanks. I will use the USB to serial converter that I have already. 

Yes, its connected to COM5 - verified in Device manager.

Well, regarding the upload_speed, this line is commented in my platform.ino file:
`; *** Upload Serial reset method for Wemos and NodeMCU`
`;upload_resetmethod = nodemcu`
`;upload_port = COM6`
`;upload_speed = 115200`

I also tried to uncomment the above line and flashed it - Did not work - Same results:

> Building .pioenvs\sonoff\firmware.bin
Configuring upload protocol...
Looking for upload port...
Auto-detected: COM5
Uploading .pioenvs\sonoff\firmware.bin
error: Failed to open COM5
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1

znanev, 2018-05-04:
These messages:

```
error: Failed to open COM5
error: espcomm_open failed
```

are saying that your COM5 port is probably used by some other process. Do you have a serial monitor (or PuTTY) keeping this port open by any chance?

Try starting your IDE (or command prompt, depending how you start esptool) with administrative privileges. If this still fails, try to establish link with your device directly from esptool, i.e.:

```
esptool --port COM5 --baud 115200 read_mac
```

Be sure to put your Sonoff T1 in serial bootloader mode before executing the command above.

oblansundar, 2018-05-05:
Yes, indeed, PUTTY was open while flashing. I just had it open for verifying the boot mode, which i forgot to close it while flashing.

After I closed putty terminal, flashing worked fine without any issues. Thanks very much.

oblansundar, 2018-05-05:
Closing this issue as it is solved.

closed 2018-05-05 by oblansundar