arendst/Tasmota #2624: compile error: /x used with no following hex digits.

consolacion, 2018-05-04:
Trying to compile on IDE 1.9 / on Windows 10. ESP board version 2.4.1

I keep getting a hoist of "/x used with no following hex digits" in several ino files such as the xdrive_interface.ino  xplg_wemohue.ino, xplg_WS2812.ino, xsns_01_counter.ino

etcetera etcetera
No idea how to remedy it

Jason2866, 2018-05-05:
Arduino IDE is not the best choice for such a complex project like tasmota...
Try Atom.io with platformio. Setup is described in the wiki.
You will surprised who easy things can be with a tool "that fits"

consolacion, 2018-05-05:
Thank you, but it is also described how to do it with the Arduino IDE and that doesn't work. Don't really feel like  having to install 2 new platforms just to get a piece of software in my Sonoff while I have no trouble getting my own software in there. So I was just signaling something that didn't work as it is supposed to.
It should either work or not in the Arduino IDE, not haphazard because 'it is complicated'.
I could still use one of the bin files and upload that, it is just that I wonder why there is a compilation error. 
Just wanted to give Tasmota a try because everybody is talking about it, but as it is, I will just stick to my own code :-)
That is no criticism on you and I thank you for taking the time to help

closed 2018-07-04 by stale[bot]