arendst/Tasmota #2626: Did I break it?

Palloquin, 2018-05-04:
Hi,

I just got a new Pow R2. of course I want to run Tasmota on there.

I hooked up my FTDI  adapter and tried to upload Tasmota. I keep getting errors though:
```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```
I opened up my FTDI adapter (there's a case around it) and it seems the I had the thing setup using the 5v output instead of 3.3v. Did I break my Sonoff? 

The sonoff still works fine, running factory firmware and using the ewlink app.

I changed the connected pin on my FTDI adapter to the one labeled 3.3. But still no luck. At no time do I see lights come on on the Sonoff during the flashing process.

Did I break my Sonoff?  or am I not succeeding in getting it in to programming mode?

rene9900, 2018-05-04:
na you did not break it, you just did not succeed in grounding gpio 0 to put it in programming mode :)

read here how to:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual-and-Dual-R2

Palloquin, 2018-05-05:
Hi Rene,

Thank you for your response!

I followed these steps perfectly at least 100 times... No luck.

It there any chance that on my board GPIO0 is not connected to the button? like on the DUAL R2?

rene9900, 2018-05-05:
I have the old pow, it has gpio0 connected to the button, i connect everything Then Press reset On fdti and hold it then Press button on pow and hold While release on fdti

Jason2866, 2018-05-05:
Ftdi adapters do often make problems. You find many issues with flashing problems.
Buy one with a CH340 chip or a CP2102 and your problems will be gone

Palloquin, 2018-05-05:
I'll try another ftdi. I have two, neither worked...

Can someone confirm the gpio0 is wired to the button on this board:
https://ibb.co/g3rM8n

Thank you all!

alexeinz, 2018-05-05:
yea 
just flashed another pow r2 this morning
it worked but takes quite a few tries for some reason with 2 ftdis too.

i can recommend two things
set baud at 115200

when connecting the usb or ground wire
make sure you hold the button and keep holding it few seconds ( if starts blinking blue , repeat the above )

Palloquin, 2018-05-05:
Got it!

What helped me out in the end was using an Arduino instead of an FTDI, as described in this video: https://www.youtube.com/watch?v=qqSLwK1DP8Q

Strangely in this case you connect RX-RX TX-TX. The labeling on the Arduino might be inverted.

Thanks all!

rene9900, 2018-05-05:
get one of these....it works fine
https://www.ebay.com/itm/USB2-0-6Pin-CH340G-Converter-for-STC-Arduino-PRO-Instead-of-CP2102-PL2303-To-TTL-/191146797242?hash=item2c813ca8ba

Jason2866, 2018-05-06:
@rene9900 
I can recommend this one too. I flashed all my sonoffs ( >10) with this one
and everytime it worked at first attempt!

Palloquin, 2018-05-07:
Thanks all!

closed 2018-05-07 by Palloquin